opentrack/opentrack #203: Center / Zero bindings aren't working

bodylinksystems, 2015-07-24:
Bound the center and zero function to various keys and either they don't work or I cannot tell if they are working as there is no visual feedback there is anything happening.

Center key toggles the apple settings menu, but numbers do not reflect being centered and zeroed.

sthalik, 2015-07-24:
Some keys may indeed not work. If you bind "Home" or "Page up" won't they not work either?

sthalik, 2015-07-24:
We're working on this in #118

sthalik, 2015-07-24:
That is, #118 will fix this issue implicitly.

sthalik, 2015-07-26:
@bodylinksystems I have a working build for Mac, want to test before rc15?

bodylinksystems, 2015-07-27:
happy to test, but this week is insanely busy.  might not be able to report back until the weekend.

sthalik, 2015-07-28:
That's understanable. Here's a current build: https://www.dropbox.com/s/5iz6nfl0hhb97uv/opentrack-osx-2.3-rc15p1-44-g7c7f4d3.zip

sthalik, 2015-08-21:
Arbitrary keys should work now also on OSX.

closed 2015-08-21 by sthalik