opentrack/opentrack #204: Diagonal movement bit like stair case instead of smooth line

MathijsG, 2015-07-24:
This program is almost perfect to me. Everything works very smooth and it keeps getting better :-) thank you!

One minor thing I still encounter, don't know if this can be better or that this is inherent to the optical tracking thing, but when I make a diagonal move, for example from right down to left up, instead of one smooth line, it's a bit of a staircase movement. Others who encounter this?

sthalik, 2015-07-24:
Accela problem. You're moving slowly and Accela only keeps up after a larger threshold. It's not limited to diagonal rotation.

Or are you talking about translation here?

Either way we need to be very careful not to make Accela worse than it is.

MathijsG, 2015-07-24:
I'm talking about a combined pitch yaw movement, for example bottom left to right up.

sthalik, 2015-07-24:
All axis in accela are treated separately. Can you test some debug builds with extra info in them?

I can probably do some guesswork as to what's wrong with Accela right now and do blind changes.

What's your accela config btw?

MathijsG, 2015-07-24:
Trying out a bit. Accela at full without the other axis seems also working.

Currently having this:

[image]

For quick responsiveness.

It's a very minor issue, but one of the few things I still encounter when playing.

sthalik, 2015-07-24:
We won't make Accela respond slower, that's for sure. There's an improvement on how it handles "small" changes. but to fix it I need to find where on rotation function the "staircase" behavior lies. See:http://goo.gl/BsTK6f

sthalik, 2015-07-24:
How many degrees before "staircase" behavior occurs can you sustain? Good guess is 1 or 1.33.

sthalik, 2015-07-24:
And by the way, is translation too jerky in rc15p1?

HMRMike, 2015-07-24:
Everything feels as smooth as ever. I did notice "staircase" sort of diagonal rotation in much earlier versions, but now it's all very smooth, with minimal filtering.

sthalik, 2015-07-25:
> What does seem too much is the limit of "snap" rotation

There was a bug where function range was limited to 2.0 and 2.6 degrees is way faster. Grab this build: https://www.dropbox.com/s/rme0cqkojkdocts/opentrack-20150725_1.7z

Please share if snap movements got better.

FlyingCircus-, 2015-07-25:
I must say I did notice the staircase effect in the latest versions. Everything is smooth and accurate, I can steady the look with a small deadzone in a way I couldn't do before, but I also noticed how the larger the deadzone the stronger this staricase effect is. It made sense because I thought that allowing finite increments of TX and TY might cause this, but the thing is you cannot get rid of it completelly even if you set the rotation deadzone to 0.

HMRMike, 2015-07-25:
>Please share if snap movements got better.

Yep! Much better. I use the tracker in DCS a lot, and sometimes must move the view fast to avoid losing track of the target. The "lag" caused by earlier versions wasn't very nice. 
Still, VERY fast movements are limited, but I doubt anyone will ever move their head so insanely fast.

sthalik, 2015-07-26:
@FlyingCircus- I'd like to run some tests. I need however a camera feed video along with model dimensions and Accela settings.

And yes, deadzone causes the staircase effect a lot. Maybe we should remove deadzone again or at least restrict it to very small values.

FlyingCircus-, 2015-07-27:
@sthalik Here you've a link to the camera feed and my OpenTrack settings: https://www.dropbox.com/sh/qnk65oh30ixh036/AACs9dyIo_xhBDlzmsFjoFuKa?dl=0

I'll prefer if you don't get rid of the deadzones, as I like to set them up with low values even if this causes a little staircase effect. What is surprising is that the effect is still slightly noticeable with 0 deadzones while I don't remember experiencing it before I came back with my new clip.

sthalik, 2015-07-28:
@FlyingCircus- the video plays really fast, I need the real speed since Accela works with time units.

sthalik, 2015-07-28:
Are you sure it's the right video? It looks like the same one you posted way before.

FlyingCircus-, 2015-07-29:
Nah, but the movements where quite the same. Those were recorded the own CL-Eye app for the PS3, which generates an odd output, but I thought that would be as raw as it could be.

I have updated the link with a video captured with a different software, hope it's more of your linking:
https://www.dropbox.com/sh/qnk65oh30ixh036/AACs9dyIo_xhBDlzmsFjoFuKa?dl=0

sthalik, 2015-07-30:
We need to delay this somewhat, my workstation has hardware issues....

FlyingCircus-, 2015-07-30:
I have no hurry at all. I'm not going to play any sim for a while, neither I've the joystick to do so. I'll follow the development anyway.

MathijsG, 2015-08-08:
For me it didn't get better in RC15. Still testing some stuff out.

For me it feels like choosing between idle jumping movement, because head never sits still and fast response, or filtering out those jerky tiny movements but causing tracking to be less responsive and slightly delayed.

sthalik, 2015-08-11:
Try adding just a little smoothing, below 20ms.

sthalik, 2015-08-11:
As for rc16 - I've got stuff to do daily during weekdays, and it limits my computer time immensely. It's also fucking hot and it makes me exhausted. There are few problems I'd like to address from this issue tracker before rc16, but can't at the moment.

cheers!

sthalik, 2015-08-30:
@MathijsG @FlyingCircus- any updates?

FlyingCircus-, 2015-08-30:
@sthalik Regarding the "staircase effect" I see no improvements, but I think that's something to expect for.
I've painted a visual representation of what I think it's happening. I believe current deadzones in Accela work as a function similar to x = Dx * floor(x/Dx), being Dx the deadzone of such x coordinate. You might be rounding instead of flooring, the effect would be the same.
The problem with such algorithm is that you deal with each coordinate independently and you're fixed to a grid such as the following:

[image]
Where the small dots represent the pixel resolution, the big dots represent the deadzone resolution, the light red arrow represents the actual movement and the dark red arrow the result of the deadzone function.

What we really need is more complex: distances (or rotations) from last location should be taken into account for the deadzone, involving all coordinates at the same time. The actual algorithm we're looking for should look something like this:
if (x2-x1)^2 + (y2-y1)^2 + (z2-z1)^2 >= D^2  ->  x1=x2, y1=y2, z1=z2  else  x1=x1, y1=y1, z1=z1
Being D the deadzone magnitude, c1 the last value of the coordinate and c2 the current actual value of the coordinate.

In such a case, we would get a behaviour such as the following:

[image]

I hope my explanation is clear enough.

sthalik, 2015-08-31:
@FlyingCircus- I don't understand how's that of any improvement.

> x = Dx * floor(x/Dx), being Dx

Formula used is max(0, x-Dx) before looking up the spline.

> if (x2-x1)^2 + (y2-y1)^2 + (z2-z1)^2 >= D^2 -> x1=x2, y1=y2, z1=z2 else x1=x1, y1=y1, z1=z1

That's just Euclidean distance over all three coordinates. It doesn't change deadzone behavior.

Graphs are bogus. The first one has the "staircase" exceed the straight line. The second doesn't follow from your equation.

FlyingCircus-, 2015-08-31:
@sthalik I believe you'll always get a "staircase" effect as long as you implement deadzones by using formulas for each coordinate that depend only on themselves (x as function of x and only x, etc).

I don't see mistakes within the graphs, but they might not be clear enough. Let's talk about translations only:

[image]

That's the beginning of the first graph, the movement of the LEDs from the webcam point of view is represented by the light red line as before. The actual pixels the webcam sees are the light red dots (depending on its resolution). When the x coordinate reaches a distance equal to the deazone from the last point it updates its position to 1, the same happens with the y coordinate when it updates it's position to 2. If you follow all along you get the complete graph. I'm not accounting for missing pixels due to refresh rate, but that wouldn't change what happens. The problem with this behaviour is that you're locked to a single axis displacement. Also, the larger the deadzone the more noticeable this effect is.

To solve this you need something as my suggestion: don't update ANY coordinate unless the Euclidean distance (as you said) from the last viable coordiante is equal or greater than the deadzone. That way you won't get single axis displacements, neither it would be worse with larger deadzones.

Anyway, that's how I think deadzones should work for translations, which are the easy case. For rotations you need to calculate the total angle variation as a consequence of the pitch-yaw-roll variations. Then we could apply the same: don't update ANY rotation axis unless the angle variation from the last viable coordinate is equal or greater than the deadzone.

Using wikipedia as source it seems we would need to compute the trace of the product of the 3 rotation matrix: https://en.wikipedia.org/wiki/Rotation_matrix ("Determining the angle" in "General rotations"). I might look further into this if you think it's worth a shot, I've to convice you first...

sthalik, 2015-08-31:
@FlyingCircus- for translation this isn't that obviously best. POSIT has large noise for the Z value and little for X and Y. It also depends how large is a "large deadzone". For Euclidean distance you need larger possible value, though.

I can make a test build, but do you have an underlying basis why it's better at all?

FlyingCircus-, 2015-08-31:
@sthalik It's better because all axes should be updated at once, either case we will get "staircase" movements depending on which one reaches first the deadzone condition. Differences between first and second graph were trying to illustrate this, it's clear the second one follows the path much closer than the first one while keeping as a condition that variations below the deadzone shouldn't be taken into account.

You're right the effect is much less noticieable with translations, but it was also much easier to illustrate. Anyway, rotations will suffer the same way.

FlyingCircus-, 2015-09-01:
According to the suggested reference in wikipedia, if we want to apply my idea of deadzones to the translations, the equation we should fulfill would be:

if  abs(arccos( (cos(Tx)*cos(Ty) + cos(Tx)*cos(Tz) + cos(Ty)*cos(Tz) - sin(Tx)*sin(Ty)*sin(Tz) - 1) / 2 ) )  >=  Dr  then update values, either case don't update them. Being Dr the translation deadzone.

I know it's a long calculation, but it might worth a try

sthalik, 2015-09-02:
@FlyingCircus- trig wraps around on an interval specific to given trig function. Say, pi/2 or 1. I don't see, really, why it's applicable here more than anything else, like <code>f(x) = x</code> or a sigmoid function.

@FlyingCircus- I've had a test build like that. I should dig it up sometime soon and give you guys for comparison.

FlyingCircus-, 2015-09-02:
Nice, if we get around this problem the filter will be much more useful to me. I never needed much filtering, I only needed a way to completely steading a pose. Introduction of deadzones achieved this but they also brought the "staircase effect" we are trying to get rid of.

sthalik, 2015-09-03:
@FlyingCircus- see if https://www.dropbox.com/s/ea0ayzbxmj07eou/opentrack-test-20150903.7z improves, thanks.

FlyingCircus-, 2015-09-03:
@sthalik I don't appreciate any differences between the linked version and the previous one. Staircase effect is still there...

FlyingCircus-, 2015-09-03:
@sthalik Looking at your commit I see translation and rotation variations are calculated independtly and applied independtly at each axis: I see indexes between 0 to 2 correspond to Tx, Ty and Tz while indexes 3 - 5 are Rx, Ry and Rz, deadzones are applied in a loop from 0 to 5 to each coordinate.
I'm suggesting in this thread that we should combine those variations (to compute the Euler total distance and total rotation from the formulas above), check if the deadzone condition is fulfilled and then decide to update all axes at once or not.

I might be interpreting the code wrong, sorry if that's the case.

sthalik, 2015-09-04:
@FlyingCircus- it's checked against the deadzone fine.

    dz*dz > (i >= 3 ? rot_dist : trans_dist)

sthalik, 2015-09-04:
@FlyingCircus- there was a difference from your formula, grab https://www.dropbox.com/s/2hzx9fkg9n5kery/opentrack-test-20150904.7z

FlyingCircus-, 2015-09-04:
@sthalik I'll test it this evening. Still there're a couple of concepts we're trying to improve here:

1.The shape of the deadzones: in case of translations that would be cubic in 3D space with the former implementation, instead of spherical.

2. The way they're applied: in the former implementation that would be independently to each axis (if one axis variation is greater than the deadzone then it gets a new value while the other axes keep the same unless they also fulfill their own deadzone condition), instead of updating all axes at once if the global deadzone condition is fulfilled.

2 has much more impact on the "staircase effect". If we check and update axes one by one in a loop such the one in the commit then we won't improve 2. I hope those last bulids are addressing this. Keep up the good work!

HornetMaX, 2015-09-04:
The way I'd summarize what FlyingCircus is saying is that (in his 2d example/plot above) given one light red dot (camera pixel), you need to pick one black dot. Which one do you pick ? The closer one.

For translations the distance is the classic euclidean ones, for rotations it's the "angle between the two attitudes" (angle of the single rotation that bring from one attitude to the other).

sthalik, 2015-09-04:
Thanks, guys.

> If we check and update axes one by one in a loop such the one in the commit then we won't improve

In the code I quoted deadzone is checked and applied once, prior to one-by-one processing.

Maybe we could make the deadzone interpolate non-linearly, i.e. exponentially? Rather than staircasing, i.e. https://en.wikipedia.org/wiki/File:Tchebyscheff5.png Can't find a better picture sadly.

sthalik, 2015-09-04:
This one https://www.dropbox.com/s/klafddfqtec3uuk/opentrack-test-20150904_2.7z scales value against deadzone linearly before applying Accela gain. There's no staircase effect, and screen is stable.

FlyingCircus-, 2015-09-04:
@sthalik Great work! That last build seems to do the trick. However, I'm setting the deadzone slidders all the way to their maximum value (degrees and mm units don't make much sense there), we just need a better scale and it'll be perfect.

FlyingCircus-, 2015-09-04:
I see what happens: now deadzones are sensitivity dependent.

sthalik, 2015-09-05:
This one <<https://www.dropbox.com/s/f9kegphi4y0ia5t/opentrack-test-20150905_1.7z>> uses deadzone size in place of sensitivity if they apply. It's in principle correct, at least. Is there improvement?

FlyingCircus-, 2015-09-05:
Yes, it seems to be fine now. There's still some room for improvement in those last builds. We don't have staircase effect anymore, deadzones get rid of most of the noise, but there's still some noise lagging behind. Let me explain how I'm getting my settings now:

1. I move all sliders to 0.
2. I move only the deadzone slider until I can keep a pose steady (first step is enough with the last build).
3. I filter slightly the output by using the sensitivity and smoothing sliders.

I get a lot of noise after 1, which is normal. I get rid of that noise after 2, depending on my deadzone settings I'm able to notice the deadzone resolution (view moves in incremental steps). I filter the incremental steps after 3.

That's the way I expect it to work. However, after setting the deadzones (step 2) I still cannot get a completelly steady view: I don't get a shaking view such as before but I get a noise-lagging behind effect.

sthalik, 2015-09-06:
No, it's bad. It destabilizes sensitivity when it's used by itself if your issues are addressed. I'll be reverting to the staircase-like version. I recommend then using no more than 0.3deg deadzone. Also, sensitivity was made less sensitive at the very start, so that view is more stable.

FlyingCircus-, 2015-09-06:
To be fair my sensitivity slider was all the way down before you were addressing the deadzones, therefore I don't really know what changed on its behaviour. Anyway, I'm surprised deadzones let some noise get through, as if the sensitivity filter was still being applied and added up to the output.

sthalik, 2015-09-06:
in latest builds deadzone wasn't absolute i.e. put up to 1 degree output to avoid aforementioned staircasing. then what remained got plus one degree to preserve continuity. it worked badly.

sthalik, 2015-09-06:
in other words, we can't solve deadzone problem like that. keep it at low values...

FlyingCircus-, 2015-09-06:
Ok, that's why I was noticing the deadzone resolution and I thought we just needed to work on the scale. My explanation for the staircase effects remains valid: we get it because one translation/rotation coordinate reaches variations larger than the deadzone before the other coordinates (which is normal) but then we only update that axis (which shouldn't be). If any coordinate has a variation larger than the deadzone ALL coordinates should get new values.

Idealy we also shouldn't check for single coordinate variations but total displacement/rotation variations.

sthalik, 2015-09-09:
That'll cause staircase on all axis, not one.

FlyingCircus-, 2015-09-09:
It was my impression we have staircase on all axis now...

To oversimplificate the problem: if we consider translations in 2D Tx-Ty, we forget about pixel resolution and refresh rate, and we even check for single coordinate variations instead of total displacement, when we make a movement such as A - B the following happens with the current system:

[image]

Tx  variation is larger than the deadzone resolution, but Ty's isn't -> we make Tx to update its value, but Ty keeps the same, we end up in B'.

What we should be looking for is the following:

Tx  variation is larger than the deadzone resolution, but Ty's isn't -> we make Tx AND Ty to update their values, we end up in B.

So instead of looping all coordinates, checking for coordinate variations and acting in consequence one by one, we should check coordinate variations for all of them (looping if you want, but idealy calculating total displacement/rotation variations) and we should act in all of them AT ONCE, aka don't update any of them unless any of them has suffered a variation larger than the deadzone (or ideally: the combined variation is larger than the deadzone).

FlyingCircus-, 2015-09-09:
Sorry if I'm being too adamant. I believe what I'm suggesting is achievable, but I might be wrong.

sthalik, 2015-09-09:
@FlyingCircus- it's fine. It'd help understanding though if you dealt with rotations/translations and not point centers.

sthalik, 2015-10-09:
@FlyingCircus- is it still a big problem on rc19. Some accela regressions got fixed.

FlyingCircus-, 2015-10-09:
I found last version behaved a bit different regarding the values of deadzone and sensitivity I should use. However staircase effect is still slightly noticeable.

I'm no programmer and I was trying to avoid quoting your code, but I guess there's no other simpler way to explain what I thought was the cause of the issue. Let's look at https://github.com/opentrack/opentrack/blob/eb7703b4ecea5540a4c4a65faa1ff4c53fe97631/ftnoir_filter_accela/ftnoir_filter_accela.cpp

Consider we replace the last "for loop" with these two:

```javascript
bool isnt_trans_dz = false;     // deadzone flag for translations
bool isnt_rot_dz = false;       // deadzone flag for rotations
for (int i = 0; i < ; i++)
{
    smoothed_input[i] = smoothed_input[i] * (1.-alpha) + input[i] * alpha;
    const double in = smoothed_input[i];
    const double dz = i >= 3 ? rot_dz : trans_dz; 

    if (i < 3)
        isnt_trans_dz = isnt_trans_dz || (fabs(vec) >= dz);     // any translation coordinate might activate the flag (ideally we should check for combined translation)
    else
        isnt_rot_dz = isnt_rot_dz || (fabs(vec) >= dz);         // any rotation coordinate might activate the flag (ideally we should check for combined rotation)
}

for (int i = 0; i < 3; i++)
{
    Map& m = i >= 3 ? rot : trans;
    const double vec = in - last_output[i];  
    const double thres = i >= 3 ? rot_t : trans_t;

    bool isnt_dz = i >= 3 ? isnt_rot_dz : isnt_trans_dz;                 // check for deadzone, it will be true of false for all translations or all rotations at the same time
    const double val = isnt_dz ? m.getValue(fabs(vec) / thres) : 0.;     // apply deadzone when suitable

    const double result = last_output[i] + (vec < 0 ? -1 : 1) * dt * val;
    const bool negp = vec < 0.;
    const bool done = negp
        ? result <= in
        : result >= in;
    const double ret = done ? in : result;
        
    last_output[i] = output[i] = ret;
}
```

This way we don't hold a single translation/rotation coordinate if it didn't have a variation greater than the deadzone, we hold all coordinates or none of them.

Sorry if my coding practices are bad, I might be doing something wrong but I'll be happy if you get the idea I'm trying to explain and you code it the right way.

sthalik, 2015-10-10:
@FlyingCircus- it was more jerky or at least not less jerky like that. The spacing between two minimal movements due to deadzone is larger and thus more visible. I've already tried your scheme in a private branch and it didn't work out.

sthalik, 2015-10-10:
Since you want it inverted - any coordinate causes others from rot/trans to activate - deadzone will need to be increased to accomodate this behavior. I urge you to try it at runtime and see how it works.

MathijsG, 2015-10-10:
To me,although it is a very minor thing for me, it still feels like this when moving diagonally:

Excuse my Paint skills:

[image]

Like little curved waves, instead of a smooth line. Maybe because the clip is either at the right side of someone or the left, so a diagonal can't be perfect? Will make a hat thing very soon.

sthalik, 2015-10-10:
Of course it can't be perfect. It's a *deadzone*. If you want a diagonal line, disable it.

FlyingCircus-, 2015-10-10:
@sthalik It's possible I didn't notice the jerkiness in the previous built because I was already using deadzones, although I could lower their values on the last one so I guess it was an improvement. I'm not very confident at playing with the source code, as I said my experience is very limited, but I could get my hands dirty if I ever get the time for doing so. By the way, if <code>m.getValue(0) != 0</code> then I should have used <code>const double val = isnt_dz ? m.getValue(fabs(vec) / thres) : m.getValue(0.);</code>

FlyingCircus-, 2015-10-10:
Also vec wasn't calculated on the first loop, that way it could have never worked out. Some stuff in the second loop, such as <code>const double vec = in - last_output[i];</code> should have been done on the first one. I see I'm terrible at this.

[19 more replies]

closed 2015-12-17 by sthalik