opentrack/opentrack #205: HT crashes immediately

HornetMaX, 2015-07-24:
With 2.3r15 and 2.3r15p1, as soon as I select HT and start, th eled on my webcam blinks a couple of times than opentrack crashes.

Don't know for 2.3r14, but it defintely worked with r13.

MaX,

sthalik, 2015-07-25:
Does it crash with camera different than PS3 Eye? Which Windows version are you using?

MathijsG, 2015-07-25:
Same problem here, I tried it when I read it here. PS3 cam here.

sthalik, 2015-07-25:
@MathijsG I need debug information. Can you install this https://sourceforge.net/projects/mingw/ with development tools? I'll give you exact commands to use.

HornetMaX, 2015-07-25:
For me it's win7 x64 ultimate, webcam is Microsoft LifeCam HD-3000.

MaX.

sthalik, 2015-08-01:
What's your CPU?

MathijsG, 2015-08-04:
For me it is i5 2500k.

Working a lot these days, so don't have many time soon.

HornetMaX, 2015-08-05:
i5 2500k for me too.

sthalik, 2015-08-05:
Maybe can be fixed easily with disabling some optimizations. Stay tuned.

sthalik, 2015-08-13:
I was able to reproduce the crash. Confirm this build works:

https://www.dropbox.com/s/83v9wxm3z7cgg00/opentrack-2.3-rc15p1-58-g87379aa.zip

MathijsG, 2015-08-13:
There are some weird cache files with a folder called %systemdrive% which also is a Path variable in Windows. Are those files needed too?

MathijsG, 2015-08-13:
Anyways with this build it IS working again :-) here.

sthalik, 2015-08-13:
The files seem to have been created by some analysis software. I was concerned there for a sec but the install directory and the files are clean.

Good that it works now.

closed 2015-08-13 by sthalik