opentrack/opentrack #207: Got issues while starting tracking with PS3 Eye cam.

cazzoo, 2015-07-26:
I recentrly installed RC15P1 and got issues while starting the tracking and having my PS3 eye cam set as default camera.
I get a popup windows that says me that the memory could not be read at instruction 0x6fcbd733 (If It could be usefull for you..).

I previously used RC4 with no issues.

sthalik, 2015-07-27:
Is this the PT tracker? It had camera issues that were only fixed very recently. Confirm this build fixes the crash https://www.dropbox.com/s/6j9cegmujiqtlgp/opentrack-20150727_1.7z if it does, we need to release another public build.

cazzoo, 2015-07-27:
Thanks for the feedback, I'll try this at lunch GMT+1 and will post results
on another message in this thread issue.

cazzoo, 2015-07-27:
Well I had some minutes to try this right now and unfortunately this recent build does not fix the issue.
The popup error is same with same memory address. :/

sthalik, 2015-07-28:
Run this build https://www.dropbox.com/s/f7scrm5vfejuyry/opentrack-console-debug-20150728_1.7z

Get it to crash and send contents of the console window that pops up along with the UI. Paste the window image or paste the text.

cazzoo, 2015-07-31:
Ok I finally get some time to spend on my PC, I downloaded the custom build you made and tried to run as I usually did.
Here is the ouput of the program : 
```
Loading "libopentrack-filter-accela.dll"
Loading "libopentrack-filter-ewma.dll"
Loading "libopentrack-filter-kalman.dll"
Loading "libopentrack-tracker-aruco.dll"
Loading "libopentrack-tracker-freepie-udp.dll"
Loading "libopentrack-tracker-hatire.dll"
Loading "libopentrack-tracker-ht.dll"
Loading "libopentrack-tracker-hydra.dll"
Loading "libopentrack-tracker-joystick.dll"
Loading "libopentrack-tracker-pt.dll"
Loading "libopentrack-tracker-rift.dll"
Loading "libopentrack-tracker-udp.dll"
Loading "libopentrack-proto-fgfs.dll"
Loading "libopentrack-proto-freetrack.dll"
Loading "libopentrack-proto-fsuipc.dll"
Loading "libopentrack-proto-simconnect.dll"
Loading "libopentrack-proto-udp.dll"
Loading "libopentrack-proto-vjoy.dll"
Loading "libopentrack-proto-win32-mouse.dll"
bundle + "opentrack-ui"
bundle + "proto-freetrack"
bundle + "HT-Tracker"
bundle + "Accela"
failed CoInitializeEx -2147417850
```

sthalik, 2015-08-01:
You're using the HT tracker and not PT. HT has some issues at the moment that cause crashes for few people.

sthalik, 2015-08-14:
HT fixed for rc16.

closed 2015-08-14 by sthalik

cazzoo, 2015-08-14:
Nice! I'll check that out at the end of the day!
Thanks

sthalik, 2015-08-14:
Note there hasn't been a binary release yet.

sthalik, 2015-08-21:
Fix released.

cazzoo, 2015-08-21:
Thanks for the feedback.
I'll test this out asap.
Thanks