opentrack/opentrack #210: Compilation issue in opentrack-2.3-rc15p1

npetrenko, 2015-07-28:
Compilation errors in opentrack-2.3-rc15p1

/home/nikita/opentrack/opentrack-opentrack-2.3-rc15p1/ftnoir_tracker_pt/camera.cpp: In member function ‘virtual void CVCamera::_set_fps()’:
/home/nikita/opentrack/opentrack-opentrack-2.3-rc15p1/ftnoir_tracker_pt/camera.cpp:127:23: error: ‘CV_CAP_PROP_FPS’ was not declared in this scope
     if (cap) cap->set(CV_CAP_PROP_FPS, cam_desired.fps);
                       ^
/home/nikita/opentrack/opentrack-opentrack-2.3-rc15p1/ftnoir_tracker_pt/camera.cpp: In member function ‘virtual void CVCamera::_set_res()’:
/home/nikita/opentrack/opentrack-opentrack-2.3-rc15p1/ftnoir_tracker_pt/camera.cpp:134:18: error: ‘CV_CAP_PROP_FRAME_WIDTH’ was not declared in this scope
         cap->set(CV_CAP_PROP_FRAME_WIDTH,  cam_desired.res_x);
                  ^
/home/nikita/opentrack/opentrack-opentrack-2.3-rc15p1/ftnoir_tracker_pt/camera.cpp:135:18: error: ‘CV_CAP_PROP_FRAME_HEIGHT’ was not declared in this scope
         cap->set(CV_CAP_PROP_FRAME_HEIGHT, cam_desired.res_y);
                  ^

sthalik, 2015-07-28:
use opencv 3.0 aka master branch

npetrenko, 2015-07-28:
My opencv version is 3.0.0

sthalik, 2015-07-28:
The constants should be there:

> #include <opencv2/highgui/highgui_c.h>
> #include "opencv2/videoio/videoio_c.h"

> enum [...]
>     CV_CAP_PROP_FRAME_HEIGHT   =4,

Are they not on your end?

Tickets can be closed and reopened, there's no need to create a new one each time responding.

npetrenko, 2015-07-28:
I have the mentioned header files. The problem disappeared after including them to camera.cpp . Thanks. I am starting to learn c++ and I dont understand what you wanted me to do with "enum ...".

npetrenko, 2015-07-28:
/home/nikita/opentrack/opentrack-opentrack-2.3-rc15p1/ftnoir_tracker_pt/point_extractor.cpp: In member function ‘std::vector<cv::Vec<float, 2> > PointExtractor::extract_points(cv::Mat&)’:
/home/nikita/opentrack/opentrack-opentrack-2.3-rc15p1/ftnoir_tracker_pt/point_extractor.cpp:163:48: error: ‘CV_RETR_LIST’ was not declared in this scope
         cv::findContours(frame_bin_, contours, CV_RETR_LIST, CV_CHAIN_APPROX_SIMPLE);
                                                ^
/home/nikita/opentrack/opentrack-opentrack-2.3-rc15p1/ftnoir_tracker_pt/point_extractor.cpp:163:62: error: ‘CV_CHAIN_APPROX_SIMPLE’ was not declared in this scope
         cv::findContours(frame_bin_, contours, CV_RETR_LIST, CV_CHAIN_APPROX_SIMPLE);

sthalik, 2015-07-29:
To be clear, you don't need to modify anything to compile opentrack.

My test build was made using opencv master from several weeks ago. There were no changes to the files where you get build errors.

However, you're far better off using the "unstable" branch. As you can see stable/2.2 bitrotted a lot.

sthalik, 2015-08-01:
Did you find the source of the problems?

npetrenko, 2015-08-01:
Unfortunately I am not able to make any progress in the nearest 10 days as I'm on vacation and my computer is 2000 kilometers away.

sthalik, 2015-08-30:
Abandoned, sadly :(

closed 2015-08-30 by sthalik

Vignesh1076, 2016-07-11:
Hi, I had similar issue. I could solve the issue with the 2 header files you asked to use. But when i see the display after changing the fps from 30 to 5 or 1. I see the same video.

sthalik, 2016-07-11:
@Vignesh1076 clone the opentrack-depends repo then have git clone the submodules. Build opencv from our repo as a static library.

sthalik, 2016-07-11:
@Vignesh1076 to be clear the right opencv branch is `fork`.