opentrack/opentrack #211: Opentrack 2.3 RC15PS3 Eye (with CL Eye driver) can't control camer settings

HerrStein, 2015-08-06:
Hi,

I'm not sure if it's problem of Opentrack and pointtracker 1.1 or if it's a problem of the CL Eye Driver, but when i try to set up the pointtracker and try to go to the camera settings dialog. The settings window will open but none of the settings can actually be influenced. 

I'm running opentrack on Windows 10 64-bit, it has admin right and I tried severs compatibility modes, (Win 8, 7, XP) but nothing helped.

Hope this issue can be fixed, because if think the PS3 Eye is widely used.

sthalik, 2015-08-07:
It can't be worked around at the moment. Use the CL-Eye program. This functionality is mostly for Logitech camera owners, which work this way, and their control panels are harder to access.

HerrStein, 2015-09-02:
hey. sorry for the delayed answer. i tested it again. when i change the settings in the CL-Eye programm the picture the camera sends to opentrack doesn't change. Whatever i change in the CL-Eye programm has no effect within opentrack. Maybe someone else using a similar setup can confirm this?

HerrStein, 2015-09-02:
Sry for the late answer. I figured out how to change the settings now. After some tests i realised that opentrack only updates the video feed if i stop and restart the tracking completely. But I still don't get the same tracking quality as i had with version 2.2. I think the main problem is the camera options in the PointTracker settings. in version 2.2 you could set the Threshold and Hysteresis seperatly with two sliders. in 2.3 the Hysteresis is remove and (at least for me) it feels that it was integrated in the Threshold slider. If i lower the Threshold the image also get's less bright and the dots get harder to track. Would it be possible to bring back the Hysteresis slider in the PointTracker settings? I think this would make it easier to get the programm working properly.

sthalik, 2015-09-03:
Both issues with camera settings are known.

It's not the hysteresis, point extraction algorithm was changed. Do the dots stop getting detected presently?

HerrStein, 2015-09-03:
they always stop being detected for splitseconds. like flickering.

sthalik, 2015-09-04:
See if this build improves - https://www.dropbox.com/s/dkupldzvajyfa6h/opentrack-pt-extraction-20150904.7z

HerrStein, 2015-09-04:
i tested around for a while and it definitely feels more stable

sthalik, 2015-09-04:
How much? Are there still problems?

closed 2015-10-09 by sthalik