opentrack/opentrack #212: New "center pose offset" behavior is motion sickness inducing! Option for old behavior?

eobet, 2015-08-08:
I can't fathom the use case for the new "center pose offset" behavior which happened somewhere between rc9 and rc15. It actually offsets not the center of your head, but your pivot point, so the effect is something like spinning on a carousel when you're looking left and right.

To put it another way: In the sims I play, I use a really low FOV which zooms me in quite a bit, and to offset this, I used to set TZ a couple of centimeters back in OpenTrack, and everything was fine. I could still look around normally from this new position. But in the latest release of OpenTrack, if I do this, my virtual head suddenly starts swinging left and right, almost crashing into my virtual cockpit.

It's simply impossible to use this anymore for me.

The effect can be seen in the main OpenTrack window, as TZ starts out at the value set in the options, but as you turn your head, the value (instead of being fixed) goes down to 0 and TX or TY starts to move, resulting in instant head-aches.

Please at least include an option for the old, fixed behavior.

Thank you!

sthalik, 2015-08-09:
Try this build, fixes #212 and #213 https://www.dropbox.com/s/plbh6vvpdg3ak9r/opentrack-2.3-rc15p1-50-g573a78e.zip

sthalik, 2015-08-09:
It first maps, then compensates, and only then applies zero offset now.

eobet, 2015-08-09:
YES!

Thank you very much, this has resolved my issue.

I can now offset and still get normal head movement.

sthalik, 2015-08-09:
Thanks for confirmation. It's included in future rc16.

closed 2015-08-09 by sthalik