opentrack/opentrack #213: X Translation axis configurable.

HellfireLx, 2015-08-08:
The new 2.3 rc15 is awesome with Arma3, the smoothness of the Accela filter with minimal settings and, hence, less lag, is really superior to the previous iteration. This is not a bug report but a feature request. As roll and yaw seems to always go together with a trackIr build device, I use X axis  for roll movement triggered behaviour. This keeps the yaw and roll based movements nicely separated. Today I am limited to a 100 units and I would need 180 as in roll, so that arma 3 does a full right or left leaning. Is there any way to make the axis max values configurable (at least in a txt file, I don't need a UI slider) .

Thanks for the great work !

sthalik, 2015-08-09:
https://www.dropbox.com/s/plbh6vvpdg3ak9r/opentrack-2.3-rc15p1-50-g573a78e.zip

sthalik, 2015-08-09:
That is, in this build remap roll -> X and treat the roll axis as arma's X axis.

HellfireLx, 2015-08-09:
Great, full leaning is now working on X axis. 

Many Thanks

closed 2015-08-09 by HellfireLx