opentrack/opentrack #215: [Hardware] Inexpesive hardware integration?

Tiktiki, 2015-08-12:
http://www.hobbyking.com/hobbyking/store/__61695__Quanum_3_Axis_No_Drift_Head_Tracker.html

Really cheap head tracker featuring 3-axis GYRO, ACC and MAG.

Would be really nice to integrate with opentrack.
It is based off an Arduino Nano board.
Outputs to USB.
Has as remote drift reset/recenter button.

I can't code for shit, it would be great to use with opentrack.

bodylinksystems, 2015-08-13:
That is fantastic. I would love to see that happen. Having my cellphone on my head is getting old.

sthalik, 2015-08-14:
If there's reasonable demand then sure.

YellowOnion, 2015-09-24:
There's 2 other arduino based projects around, EDTracker outputs to a basic HID-USB (joystick) device (no special support required), and then the hatire plugin is used for some other method (serial over USB perhaps?).

Since this project looks like it's based on a non-USB compatible chip, it's probably using serial over USB, the project is meant for people who want to send the data via RF, to some device (drone with movable camera?), so not ideal for head tracking in games, and since serial is too generic I don't think hatire will work with this.

So on the off chance someone is considering buying one of these, I would recommend the EDTracker DIY project instead ($15USD~ on ebay), or the ED Tracker Pro (though double that of the Quanum). both have a full GUI setup.

sthalik, 2015-10-09:
Or use a phone with a gyro.

closed 2015-10-30 by sthalik

bodylinksystems, 2015-11-10:
I would like to use a different device other than my phone on my head.  Perhaps I buy the edtracker kit for $15?  Will it work with opentrack?

sthalik, 2015-11-10:
EDTracker uses a proprietary sensor array. They can't even change the code that computes the pose. I'd like to work on Arduino-based firmware but for that I need someone available in real time. The way they compute the pose is really damn naive.

HornetMaX, 2015-11-10:
I have an EDTracker (Pro, but doesn't make any difference) and it works perfectly with opentrack as EDTracker is seen as a joystick so you can use "Joystick input".
I have no big experience in head-tracking devices but on the forum of the game I'm in (GP Bikes) many others (with experice in HT devices for Flight Sims) said that EDTracker works much much better than TrackIR for example. Very stable tracking, I don't even need a filter in opentrack.

bodylinksystems, 2015-11-11:
I'm interested in working on this, but can only test. I havent written any processing stuff in a long time