opentrack/opentrack #216: Rolling in-game always moves me off-center

MathijsG, 2015-08-13:
Always when I want to roll the view also goes to the left way or the right way (according to which direction I'm rolling to)

Do other people encounter this too? Or does everyone just have a perfect roll?

I will post a video of it soon.

sthalik, 2015-08-14:
Is there any difference with and without translation compensation?

MathijsG, 2015-08-14:
I will try this later today.

One of the problems is this: because normally both axes Yaw and pitch have a range of 180 degrees in-game, so if you make a roll and you are moving slightly off then the roll is way off centre. Maybe something that recognises doing a roll and then ignoring the other axes would be a solution too?

sthalik, 2015-08-14:
What if you yaw and roll a little? Roll gets applied, yaw gets ignored.

sthalik, 2015-08-16:
I have some unfinished work with focal length calculation for PT and aruco. Maybe it'll help.

MathijsG, 2015-08-16:
As a noob the point to mee seems that the sensitivity of yaw makes it off centre very rapidly. I will investigate further. I didn't really notice real difference between translation compensation on and off. Is that possible?

cossacksman, 2015-08-18:
I also experienced this as well as a friend as we were trying out the project using gyroscopes (ridiculously, using our phones until we can etch out some PCB's for a small gyro board). I managed to sort it with some simple deadzones but it was apparent that I was not yawing as I rolled, though I had a similar issue with pitching.

sthalik, 2015-08-18:
What do raw values say?

Does it happen if you don't center at all?

sthalik, 2015-08-30:
@cossacksman @MathijsG try this build https://github.com/opentrack/opentrack/issues/227#issuecomment-135718763

sthalik, 2015-10-02:
See #237

closed 2015-10-02 by sthalik