opentrack/opentrack #218: Auto-reverting Options

cossacksman, 2015-08-18:
Using the interfaces available for editing options (Settings ports, IP's, key bindings, etc.) results in the settings disappearing as soon as you hit ok; the input tracking still works after changing my IP even though the options clearly state otherwise but the "options" menu and even the curve mapping just reverts to default if I click OK and then open the options again. The only thing the console gives me (I've added OpenTrack.bin to my PATH for launch so console is my method of execution here) is this;

```
QObject: Cannot create children for a parent that is in a different thread.
(Parent is QUdpSocket(0x28aa008), parent's thread is QThread(0x2245170), current thread is QThread(0x28a9fc8)
```

cossacksman, 2015-08-18:
This seems to be caused by the default.ini file being empty upon my build, I'm not sure if this was just my successful build or if this has happened to others but copying in a populated configuration file seems to fix this; I'd previously had OpenTrack installed on Windows but I spend the majority of my time in Linux so this wasn't much of an issue for me (anymore).

sthalik, 2015-08-18:
Did you have write permissions to the .ini? If it was empty after you switched trackers, it should populate itself after you had exited the software.

cossacksman, 2015-08-18:
Correct, apologies for the mistake, a lack of sleep and documentation for OpenGL has kept me up all night on a project!

sthalik, 2015-08-18:
What was the cause? Whether file permissions or otherwise it needs not to happen so please describe your solution.

sthalik, 2015-08-19:
@cossacksman please get back to me on it when you can.

sthalik, 2015-08-24:
Now options are stored in ~/Documents/opentrack-version, as well as the error of the inability of getting write access to the file popping up way more often, like when changing trackers or protocols. If someone sets perms to 000 or chowns ~/Documents badly, we can't do much to make it work even for the current session.

cossacksman, 2015-08-26:
Apologies, I didn't see the reply here; the cause was most definitely a file permission problem when executing for the first time, I just gave the user rights with chown and all was good, my fault for placing the executable somewhere obscure to be honest! Wouldn't really consider it a bug, loving your work so far

cossacksman, 2015-08-26:
As far as I know launching the executable file from part of the file system the user already has permissions to should yield proper results but since I compiled the binaries myself it's my own mistake really

sthalik, 2015-08-26:
We're now using ~/Documents/opentrack-2.3 as hardcoded storage location. It should work unless broken on purpose I guess.

cossacksman, 2015-08-26:
Sounds good, the Linux support is greatly appreciated; quite nice to see the community speak up about it

sthalik, 2015-08-28:
We're probably good here. Reopen if it still sucks.

closed 2015-08-28 by sthalik