opentrack/opentrack #219: Settings automatically saved and overwritten

MathijsG, 2015-08-18:
When you change a setting to your profile. And you quit the program, the changed setting is still in the profile, even if you reload the profile file manually. So it gets saved after quitting the program without a notice. This is unwanted behaviour in my opinion. Maybe adding a prompt to save or reject the settings?

While testing out the software I sometimes lose my specific settings this way.

sthalik, 2015-08-18:
No, this is prelude to #179. The "save" button will go away, and profiles will be stored in a designated directory like "Documents/opentrack-2.3".

closed 2015-08-18 by sthalik

MathijsG, 2015-08-18:
But how can you easily reject a change without manually undoing the change then?

sthalik, 2015-08-18:
If you click "cancel" the change will be undone. As for the main software's UI, there's only tracker/filter/protocol selection there.

I don't think you need to keep a history of changes. I don't know of software that does that for its configuration. Maybe use git for your .ini files? Just a random suggestion.