opentrack/opentrack #220: opentrack-2.2 stable compilation fails

vitaminx, 2015-08-18:
Hi,

I'm trying to compile opentrack-2.2-stable-20141113 on Debian Linux.
"cmake ." runs fine:

```
-- The C compiler identification is GNU 4.9.3
-- The CXX compiler identification is GNU 4.9.3
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.28") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/roman/opentrack/opentrack-opentrack-2.2-stable-20141113
```

But "make" fails:

```
[...]
[ 16%] Generating ui_ftnoir_curves.h
[ 17%] Generating ui_ftnoir_keyboardshortcuts.h
[ 18%] Generating ui_facetracknoir.h
Scanning dependencies of target opentrack
[ 19%] Building CXX object CMakeFiles/opentrack.dir/facetracknoir/global-shortcuts.cpp.o
[ 20%] Building CXX object CMakeFiles/opentrack.dir/facetracknoir/tracker.cpp.o
[ 21%] Building CXX object CMakeFiles/opentrack.dir/facetracknoir/plugin-support.cpp.o
/home/roman/opentrack/opentrack-opentrack-2.2-stable-20141113/facetracknoir/plugin-support.cpp: In constructor ‘DynamicLibrary::DynamicLibrary(const QString&)’:
/home/roman/opentrack/opentrack-opentrack-2.2-stable-20141113/facetracknoir/plugin-support.cpp:93:25: error: ‘QFile’ has not been declared
     QByteArray latin1 = QFile::encodeName(filename);
                         ^
CMakeFiles/opentrack.dir/build.make:125: recipe for target 'CMakeFiles/opentrack.dir/facetracknoir/plugin-support.cpp.o' failed
make[2]: *** [CMakeFiles/opentrack.dir/facetracknoir/plugin-support.cpp.o] Error 1
CMakeFiles/Makefile2:63: recipe for target 'CMakeFiles/opentrack.dir/all' failed
make[1]: *** [CMakeFiles/opentrack.dir/all] Error 2
Makefile:116: recipe for target 'all' failed
make: *** [all] Error 2
```

Thanks for fixing this.

sthalik, 2015-08-18:
At the very beginning of the file:

#include "plugin-support.h"
#include <QCoreApplication>
#include <**QFile***>

There's something wrong with your Qt 5 installation.

vitaminx, 2015-08-18:
never mind, one of the later 2.3-rc10 compiles fine for me (any later release uses opencv 3 libraries which are not available yet for Debian).

closed 2015-08-18 by vitaminx

sthalik, 2015-08-18:
Good to know. Also for 2.2 don't compile the tag, compile stable/2.2.

vitaminx, 2015-08-18:
stable/2.2 branch compiles fine, thanks :)

sthalik, 2015-08-19:
ok, i made a new tag for stable.