opentrack/opentrack #222: Global shortcuts not working

Gneu, 2015-08-21:
Hi all

I am facing an issue with opentrack, where Global shortcuts do not work (whatever I am in game or outside). For instance, whatever key binding added to center, when pushing corresponding key it does not center the tracker.
I am running it on Windows 10 64 bits

Many thks

sthalik, 2015-08-21:
If you run a game as admin, run opentrack as admin.

If you don't run the game as admin, don't run opentrack as admin.

Gneu, 2015-08-21:
Hi! Thanks for the quick answer! Apparently it is linked to discrepancies between AZERTY and QWERTY keyboards, as input is registered with my setup (AZERTY) but opentrack is waiting for the QWERTY input

sthalik, 2015-08-21:
Does it happen without the shift modifier?

Gneu, 2015-08-22:
I am not sure what you mean by shift modifier, but if I shift my keyboard to QWERTY then it is fine.
Another exampl is, when in AZERTY, using / or * keys on numpad do not work then after assignement. However Fx keys works perfectly (as they are common for QWERTY or AZERTY keyboards)

sthalik, 2015-08-22:
Shift modifier - the shift key on the keyboard.

I have a solution, give it some time though.

Gneu, 2015-08-22:
No problem, I have assigned function keys and it works perfectly, it is just fyi :)

sthalik, 2015-08-22:
It's not fine, we need to fix it.

Gneu, 2015-08-22:
No pb, I just meant that there is nothing urgent, at least for my case.
Thanks for all the work!

sthalik, 2015-08-30:
Note to self, use DirectInput for Windows keybindings. Not Qt signals, as done now.

sthalik, 2015-10-02:
Done, here's a test build: https://www.dropbox.com/s/avlzpywpx39o1r0/opentrack-test-20151002_1.7z

closed 2015-10-02 by sthalik