opentrack/opentrack #225: Ignore some part of camera input?

frost555, 2015-08-22:
Hello. I'm sitting near the window. Sometimes the sun is bright and i have the following situation:

[image]

Even if i rotate my head 90 degrees, my leds are not in the big bright area. Therefore it would be useful to be able to ignore some part of video input. Adjusting min-max sizes of points does't help.

What do you think?

sthalik, 2015-08-22:
Have you tried setting max point size?

frost555, 2015-08-22:
Yes, but there is different size noise at top of the my window projection on curtain.

frost555, 2015-08-22:
Right now i can try to stick some tape to camera and block desired part of image. Would like to make in software )))

sthalik, 2015-08-22:
Points are relatively small, you have much room for blocking out the noise this way.

Blue overlay isn't a problem, just make sure a cross doesn't appear there.

frost555, 2015-08-22:
Even if the sunlight is almost uniform - adjusting point size does not help
[image]
The best way is to ramp up threshold and hope that your leds are brighter than sunlight
[image]

sthalik, 2015-08-22:
On the first image no LEDs are detected at all. You should get 4 crosses.

frost555, 2015-08-22:
Yes, my bad, point are bigger than 10 pixels on that threshold level. Still, tracking is unstable and i recommend ramping threshold if you can.

sthalik, 2015-08-22:
Why is it unstable? The left side of the image should be completely ignored. It's hella big.

frost555, 2015-08-22:
Actually it is stable ingame and is smoother than with high threshold, but on opentrack window one of my leds constantly blinks. Sticked a tape, and it works even better! )) Now i can use low threshold and not worry about noise canceling effectiveness. 
[image]

sthalik, 2015-08-22:
That doesn't help. Can't we get it done without that kinda crappy solutions?

If your LED blinks then it's either threshold, min or max size.

frost555, 2015-08-22:
I dont know why it blinks, i think my thresholds are ok because i made small experiment.
Both screenshots are made on the same settings. On the second one sunlight is physically blocked.
[image]
[image]

frost555, 2015-08-22:
Is introducing virtual tape (ignoring areas on video input) is also crappy solution?

sthalik, 2015-08-22:
"Virtual tape" won't be implemented. We can find a solution other way for sure.

Put min size to 0, put max size to 20. Put theshold lower. The sunlight thing is so big it won't catch on 20+ even.

The point size may be detected incorrectly. For debugging we may also want to display point size in the camera feed.

You need lower thresholding since the point is likely contoured on one pass fine but not on another. So it's borderline thresholded fine.

sthalik, 2015-08-22:
There's also another solution but I'll skip it for now.

frost555, 2015-08-22:
Points are not detected on those settings
[image]

sthalik, 2015-08-22:
Try this build:

https://www.dropbox.com/s/e75jk2o9ywuac5q/opentrack-test-pt-extraction.7z

and tell me if the percentage points shown on non-detected blobs change anything from 100.

sthalik, 2015-08-22:
I've updated the link to include also point area.

frost555, 2015-08-22:
Yes, when i introduce sunlight the percentage drops to about 20%, blocking sunlight make them go 100% again
[image]

frost555, 2015-08-22:
Another screenshot when the sun border is more smooth
[image]

sthalik, 2015-08-22:
if the px value is close to your max size, increase it. Otherwise decrease threshold, and repeat the max size thing. Does it help?

You probably have auto exposure/gain set. Disable it.

sthalik, 2015-08-22:
Also the build has wrong px values formula, I'll send a new one.

https://www.dropbox.com/s/e75jk2o9ywuac5q/opentrack-test-pt-extraction.7z

frost555, 2015-08-22:
Still same results, but i think i'm quite ok with clunky physical tape on bright days, it happens not so often.

sthalik, 2015-08-22:
You can also rotate the camera a bit and specify camera yaw in settings.

MathijsG, 2015-08-22:
I sometimes also encounter this, with reflections in my room which acts as an extra dot in the window, killing the actual tracking. Sometimes the size of that reflection isn't really different than the regular leds, so blocking out is much harder. Maybe an idea to implement something that ignores the next encountered light source after it already has 3 leds? Or is that not a good option?

@frost555 I think you should make a better light blocker if you can still see yourself. What webcam do you use? My PS3 cam with floppy-disk behind the lense is almost black and I certainly don't see myself in the webcam image.

Len62, 2015-08-22:
Some of these issues may be related to how your PS3 cam is set-up. I had to tinker for a while before I settled on video settings in the CLeye driver.  These settings have a BIG impact on how the cam deals with the sun issue. Here is a pic of the settings I use.

[image]

sthalik, 2015-08-23:
@MathijsG we could ignore some light point, but the issue is knowing which is which, still.

petsild, 2015-08-23:
Hi, the value and quality of the points in the graphics :) facilitate the accurate adjustment of the thanks, will be included in the next rc? 
Wanted to ask if he would go at the loss of the quality of the point of the warning acoustic signal selectable on/off quiet nerušivé beep?

sthalik, 2015-08-23:
Yes, it's included in the repo.

Quality means the point is ignored for tracking. It has to be 100%. Other stuff like bright background is also shown as a point with "quality". So we'd beep all the time.

MathijsG, 2015-08-23:
For what it's worth, here my settings:

[image]

In winter time I get at the end of the day with the sun lowering big spots of sun in my filter which I can't block. Sometimes I use bedsheets to block this :-).

sthalik, 2015-08-23:
What if you rotate the camera so the sun isn't in the frame, then specify camera yaw in settings?

sthalik, 2015-09-04:
@petsild this build <<https://www.dropbox.com/s/hnfb6vmp30z2aq8/opentrack-pt-extraction-20150904_2.7z>> only uses 3 points with the most circularity. On the image you posted the left part is detected as non-circular at all.

petsild, 2015-09-04:
Hi, this adjustment has resulted in jerky movements and I had to shrink the value of the point from the original 7px to 3px and trenshold in the left. The older version worked great.

Len62, 2015-09-10:
I noticed a big difference in PT btwn rc17 and rc18. In 18 I had to adjust the cleye dr exposure and you can see how the LED's are being read now. Doesn't seem to effect it too much but it would be nice to get the values back to the levels in 17.

[image]
[image]

sthalik, 2015-09-10:
The previous values were computed incorrectly. There's no functional change here.

Len62, 2015-09-10:
Ok. Thanks for the reply.

sthalik, 2015-09-20:
@Len62 these circularity values are interesting. Please tell me if they're much lower when looking at a more extreme angle. We should get automatic threshold soon.

Len62, 2015-09-21:
They will drop to 0.50 - 0.60 at extreme angles. They are always fluctuating and it does not affect performance. I recently blocked out the source of partial sunlight I was getting at times so I re-set the exp on my eye cam. I can try and do a short video it you like.

sthalik, 2015-10-06:
@Len62 what's the circularity of the light source when it's detected?

Len62, 2015-10-06:
Using a 3 pt clip from top to bottom, at start-up:
5.2 px
4.8 px
4.7 px

sthalik, 2015-10-09:
@Len62 there's also a circularity value between 0 and 1. Show this one instead.

sniperwolfpk5, 2015-10-09:
Please give option to turn on/off these values

sthalik, 2015-10-09:
@sniperwolfpk5 what values?

Len62, 2015-10-09:
On all 3 points the value fluctuates between .75 and 1. Most of the time they stay in the .80 - .90 range. Should they be more stable? They continuously jump around.

sthalik, 2015-10-09:
When you rotate the model the points become less circular because of perspective distortion. If they fluctuate when still it could be not an optimal threshold. Have you tried new auto-threshold?

Len62, 2015-10-09:
Yes, it's about the same either way. I'll do some more testing today with other settings for my camera and see what happens.

sniperwolfpk5, 2015-10-09:
@sthalik I mean the values that are shown in Video portion beside each point.
I want to congratulate you people for the efforts that you have done. Now it is stable and i shifted from facetrackIR to opentrack.  

Thank you very much

sthalik, 2015-10-09:
@sniperwolfpk5 no, the values help me diagnose user issues. If the values obscure something, the offset can be adjusted.

@Len62 try looking at the blue portions, meaning pixels that pass threshold. Do they flicker a lot?

Len62, 2015-10-09:
No flicker in the blue areas. I managed to get the circularity to stay in the .90+ range when centered, made some adjustments to my camera.

sniperwolfpk5, 2015-10-09:
@sthalik It would be nice all the new features are documented. So it would be easier for the end user to understand new features and their effectiveness

sthalik, 2015-10-09:
@sniperwolfpk5 they are in release notes, and commit log. We're gonna have one whoop-ass of a change log once 2.3 final gets released. We're actually getting closer to release - there are issues closed and few opened, and very few new features added now. Only stuff that remains is just few long-overdue features, and some regressions to fix. Maybe 2.3 will be worthy of calling it a 'stable release'.

closed 2015-10-30 by sthalik