opentrack/opentrack #227: roll and pitch mix and effect each other despite perfectly fine raw tracking.

YellowOnion, 2015-08-24:
Why does my raw pitch seem to at random effect the output of roll and vica versa?

This is extremely frustrating as the raw numbers are not showing any anomaly.

there is some drift on yaw, which seems to effect the pitch and roll exchange amount.

I have a "center" shortcut set, but this doesn't fix the intermix of the two.

sthalik, 2015-08-24:
Centering takes all rotation into account. You have all of yaw-pitch-roll non-zero when centering.

YellowOnion, 2015-08-24:
What I mean is that I was playing fine and put my headphones down, came back and it had drifted, so I hit center, and now my roll was effecting my pitch more than my roll.

How am I supposed to fix this issue? the only thing that I can think of is having my entire desk on a swivel which is obviously stupidly impractical.

sthalik, 2015-08-26:
I need the raw pose used for centering.

YellowOnion, 2015-08-27:
Here's an example of the issue I'm facing, The only logic I can deduce is that yaw is rotating the roll/pitch around, there's no way to calibrate the phone app, so that 0 is facing my screen.

[image]

[image]

sthalik, 2015-08-28:
I don't see the problem. Raw tracker data isn't rotated. It's the tracker that inputs +-32 roll.

If it's Arduino we need to fix the firmware as part of opentrack project, but there are no other programmer volunteers than me. The current algorithm for Arduino pose estimation is hopeless. Go figure...

YellowOnion, 2015-08-28:
I don't know how much clearer I can make this, 32 roll should output something other than 0 roll should it not? when I roll my device, I want to roll, not pitch. when I pitch I want it to pitch, not roll, the raw data for roll and pitch is acting how I want it to, but the app, is not outputting to the game the way I want it to.

By rotated, I mean that the input data has been rotated around the north/south axis.

I'm using the Android app as I said earlier, not arduino.

0-yaw right now is pointing west, earlier it was pointing north, and there is absolutely no way for me to get the app to consistently point 0 towards my monitor.

sthalik, 2015-08-28:
Do you use axis remap? Also, is there any roll on the mapping widget?

Go to tracker settings, swap yaw/pitch in axis order and set pitch to minus 90. This should help you.

YellowOnion, 2015-08-28:
> Do you use axis remap?
No

I've tried swapping yaw and pitch, this only changes the amount of mixing happening, the underlying issue is not resolved, -90 to pitch doesn't help at all.

yaw seems to control the mix amount, for example if I center at 45,0,0, setting mapping 1:1 on all axis, If I move device to raw 45,14,0, it outputs 0,-7,7

sthalik, 2015-08-28:
Try this test build: https://www.dropbox.com/s/w9efehyj5ezib89/opentrack-2.3-rc17p2-1-gc5463fc.zip

Only a single line of code was changed.

YellowOnion, 2015-08-28:
Wow, that seems to have fixed it, sweet!

closed 2015-08-28 by sthalik

kuplung, 2016-01-13:
Hello,

would it be possible, to publish the above mentioned test version again? The link seems broken.

I have exactly the same problem : yaw is OK, but pitch and roll interfere each other.
The raw tracker data from phone is OK, so if i roll, then i see only roll data changing, but the game data shows not only roll, but pitch changes, and vice versa. 

Using different method does not really help, it is different, but still bad :) I tried it on two different PCs with 3 different phones, the results are the same.

Thank you!

sthalik, 2016-01-15:
@kuplung select a rotation offset in the freepie tracker options. That'll subtract or add a given value to the Euler angles.

kuplung, 2016-01-16:
Thank you the reply. Rotaion offset did not help, but it seems I found something related to the issue. 

If I lie the phone on a flat surface (my desk), and start freepie like this, then opentrack, then all 3 pith, roll and yaw values are correctly interpreted by opentrack if I start to turn the phone into different angles.

But, if I pick up the phone, like I would use in a google cardboard on my head, and re-center, then roll starts to be interfered by pitch. That means, if I look down, then after a given angle, opentrack starts to pick up too many wrong values and my look will start to roll, like I would roll my head.

In that case I need to recenter the whole thing on my desk, perfectly aligned horizontal, then everything will get back to normal. But I is rather inpractical :)

So it seems like a re-centering problem. I use the "absolute (camera)" centering method, because the other method makes e.g. Elite Dangerous completely crazy, every angle gets mixed up, and the game starts to frame drop, so it is useless.

sthalik, 2016-01-16:
If the phone is lying on a desk, the pitch value is close to zero. If you pitch up the phone, the pitch is close to 90 or -90. Then use pitch offset -90 or 90 respectively. If picking up the phone control another axis, replace that instead of pitch in the instruction.

YellowOnion, 2016-01-17:
I still have the files on my drive of the custom build:

https://drive.google.com/open?id=0BwqPn9gzXGTvTURRd1ZpcDVaLWM

You guys look like you're having the same extract back and forth as I was having.

kuplung, 2016-01-17:
Hello YellowOnion,

thanks for your reply!

I downloaded this version, I tried it. 

Now the handling of different angles seems to be correct, but the angles are mixed!

For example if I start freePIE IMU on the phone aligned with the longer egde on the desk (in a position like it would be  attached to my head in a VR headset), and I start opentrack as well, then I see following:

Raw tracker data : yaw -90, pitch 0, roll -90 -> this is translated to game data : yaw 0 pitch 0 roll 0

In that  position, if I turn the phone like I would look down in a game, then I see the roll values changing.
But, in the game this movement is translated like I would roll my head left and right.

So what I did now is the following :
- added +90 to yaw and roll in the freePIE tracker settings
- adjusted axis order : output yaw -> input yaw, output pitch -> input roll, output roll -> input pitch
- in the options/output/output remap enabled invert on pitch

With this configuration the ingame change of view is perfect.

As a remark for the developers : you do a really good job ,thank you very much.

But... Could you maybe add a bit user friendlier GUI to Opentrack? For example a graphical interface, where the user is able to select the orientation of the phone, so the above settings are made automatically.

kuplung, 2016-01-17:
Additional note to anyone who has similar problems : the above settings helped me for opentrack-2.3-rcp7 as well. 
The only difference is, that in the options menu, the centering method should be set to "relative (interial device)".

After that, everything is fine! :)