opentrack/opentrack #228: GNU GPL in opentrack project

sthalik, 2015-08-26:
Blocked by #226.

sthalik, 2015-10-11:
See 9322afafb94e4a0caffa9006ead6c19dd8f59848

Original commit at http://sourceforge.net/p/facetracknoir/codegit/commit_browser
The revision is 579d2f

@phOb what do you think?

closed 2015-10-11 by sthalik

ph0b, 2015-10-11:
@sthalik that looks perfect to me. Thanks to you and Wim!

I'll be able to improve the RS tracker a bit.

We still need to ensure we're not combining GPL modules with other incompatible ones.

sthalik, 2015-10-11:
No GPL modules remain though. Wim went over all of them.

sthalik, 2015-10-12:
@ph0b maybe we can just disallow copyleft in-tree as policy or something.

ph0b, 2015-10-12:
good. I've added [CONTRIBUTING.md](https://github.com/opentrack/opentrack/blob/unstable/CONTRIBUTING.md) to ensure this.

sthalik, 2015-10-12:
@ph0b are you fine with potential LGPL dependencies in people's projects?

ph0b, 2015-10-12:
@sthalik yes as long as there is no static builds including the LGPL software or modifications of the LGPL code.

sthalik, 2015-10-12:
can you add the bit about lgpl static linking to docs? edit interface won't work on mobile.

ph0b, 2015-10-12:
done.