opentrack/opentrack #229: [Feature] Make an official installer

MathijsG, 2015-08-26:
Now the program is evolving and becoming more and more polished with every release it may be is time for a regular installer so people can install this program officially the regular way like every other program.

sthalik, 2015-08-28:
Definitely for 2.3 final we'll provide both an installer and an archive. We already have an app bundle for OSX though.

Personally I find bloated installers annoying, with updating metadata for a minute, installing MSVC CRT redist even though they're installed already, and so on. So this is a breath of fresh air, with stuff to extract wherever you want.

We're still kinda away from 2.3 point release, all blocker bugs need to be finished before that.

sthalik, 2015-10-06:
We're good.

closed 2015-10-06 by sthalik