opentrack/opentrack #230: Identically Named Joystick Inputs - Can't Select Between Them

nanospork, 2015-08-27:
Hello,

I'm attempting to use OpenTrack to enhance my experience with an EDTracker headtracking unit. I have also built my own custom game controller using an Arduino Leonardo.

Unfortunately, while the devices show up under different names in the Control Panel (Windows 7), they have the same name under Game Controller Settings (Both show up as "EDTracker EDTracker 2"). I imagine this is because they are both based on the Arduino Leonardo board, and the EDTracker somehow set the name for Leonardo-based game controllers.

This isn't normally a problem, as Flight Simulator and other games still recognize them as unique devices. However, OpenTrack apparently cannot tell them apart, and always tries to take input from the custom game controller instead of my actual head tracker.

I have tried hacking a different name onto the controller, but it seems to be a lost cause.

So, is there a chance we can get an update which would allow OpenTrack to distinguish between identically-named devices?

sthalik, 2015-08-28:
It can be fixed rather easily. I don't reject the issue, but it's not critical.

You can plug in the desired device first so it's always chosen as a workaround.

nanospork, 2015-08-28:
Thank you for the quick response! I understand it's not critical, so I appreciate that you have at least considered it.

If it helps with development, I found that plugging in the desired device first does seem to work, but ONLY if I press "Start" before plugging in the other device. If I do so afterward, or if I press "Stop" for any reason, it seems to always detect the incorrect device when I press "Start" again.

Regardless, thank you very much for the workaround; I will finally be able to use both devices! And thank you for maintaining OpenTrack, it is an excellent piece of software :)

sthalik, 2015-08-30:
Here's a test build, confirm it works.

https://www.dropbox.com/s/f4il0qqet0o3ml4/opentrack-test-joystick-guid-20150830.7z

nanospork, 2015-08-30:
Hey, thanks for putting together a test build so quickly!

Unfortunately, in the test build, I'm still getting zero readings. I think it is still selecting the incorrect joystick (as opposed to some other issue) based on the way the GUI is behaving. I noticed that after selecting a joystick, if you click on the selection dialog again, the GUI aligns the currently selected joystick with the word Device. This is what I mean:
[image]

However, no matter which EDTracker EDTracker2 listing I click on, it always aligns as in the right-hand side of photo, with the middle one apparently selected.

This is a wild guess, but perhaps it *is* using the unique GUIDs, but it is still finding the incorrect stick in the selection process? Or, maybe there is another area in the code where it attempts to select the stick, and it's overwriting the proper GUID?

sthalik, 2015-08-30:
It's overwriting the previous GUID when you select it again. Try this one though:

https://www.dropbox.com/s/31a8vbiyxbgtx41/opentrack-test-joystick-guid-20150830_2.7z

nanospork, 2015-08-30:
The new build appears to work beautifully! Thank you so much!

closed 2015-08-30 by sthalik

nanospork, 2015-08-30:
One more note for you: the current build seems to be more jittery than the previous version I most often used. No matter how I configure Accela, there is very slight, minute jittering in FSX whenever I hold my head still. I did not experience this in 2.3-rc15p1. If it helps, the jittering feels almost "digital". Like it's jumping up and down very slightly without any precision inbetween. It's as if the map of possible angles my head can be at is too low of a resolution, so instead of moving ever so slightly it jumps, creating the jittering sensation.

sthalik, 2015-08-31:
There were no Accela changes from rc15.

The angle map has enough sub-degree resolution. It's the gain applied. Try adding a 0.2 degree deadzone or 20ms ewma smoothing.

nanospork, 2015-08-31:
I tried the 0.2 degree deadzone with 20ms smoothing in Accela, still jitters. More aggressive settings still jitter as well. The EWMA filter doesn't seem to work in Flight Simulator X, it forces a weird top-down view.

Going back and using rc15, it's still smooth, so I'm assuming it's not an issue with my hardware. If the gain is the issue, can that be changed in the settings file?

sthalik, 2015-08-31:
Maybe the noise you get is translation and not rotation? It's hard to tell sometimes.

sthalik, 2015-08-31:
Gain you control with the accela slider.

nanospork, 2015-08-31:
It could be, but my inputs are set for X,Y,Z to be disabled as my headset has no translational output. I have also disabled XYZ mapping in the Options->Output dialogue, and it seems like there may be a slight improvement but it might be placebo.

With the filter settings you described earlier, it seems more like stuttering than jittering. As if it's not updating enough times per second, or as if it's lagging for some reason.

It's curious that you said Accela had no changes, but using the exact same Accela settings (2.5ms, 1.24deg, 0deg) I get smooth action in rc15 and jitters in rc17. Is there anything else that may have changed that affects the smoothness of the output?

sthalik, 2015-08-31:
After looking again there was a single change in spline coordinates, but if anything it should improve responsiveness or minor movement.

If the beginning of the spline grows too slowly it's gonna result in zigzag movements, so it needs careful balancing.

Tracking is updated 250 times a second. You may really want to increase deadzone to less than 0.4. Also low EWMA like 10-20ms won't hurt.

nanospork, 2015-08-31:
Tried 20ms, 1.24deg, 0.4 deadzone. I'm using linear curves in the Mapping dialog. Not much improvement. It moves around smoothly, it's pretty much just when I stop moving that it stutters before coming to a stop. With the deadzone it usually DOES come to a stop eventually, but only after the stuttering. In contrast, rc15 would sometimes "float around" a bit when I stopped moving my head, but that was much less obnoxious than the stuttering.

sthalik, 2015-08-31:
Please open another issue for that.

sthalik, 2015-08-31:
We need to fix that but in its own issue for more community participation.

commit 4ad662d6 (2015-08-30): win32-joystick: fix duplicate device name selection logic

Tested-by: @nanospork
Closes #230