opentrack/opentrack #233: Translation Compensation rotation offset

kissklas, 2015-09-03:
Translation Compensation seem to use the wrong position as a pivot for rotation.It works when centered to zero. But moving again, and then rotating makes the issue reappear.

Could it be that it uses the zero position instead of the last known position, as pivot on one or more of the axes when compensation is enabled?

sthalik, 2015-09-04:
It's meant to translate depending where rotation points to. I see no issue here, provide more details.

kissklas, 2015-09-10:
Good evening, sthalik! 

Finally got around to making a recording with Translation compensation on and off. In these videos, I only rotate my head, and move back and forth.

Here is the one with compensation enabled. As you see when the head is moving away from centered position, the rotation pivot does not move with the head position, resulting in wide rotations around the centered position:
https://youtu.be/V_u099NOj_Y

Here is the same movements without translation compensation. All rotations seem correct, no matter where I move the head (The only problem of course, is that I don't have compensation when looking backwards.):
https://youtu.be/VMTh9E6Y9Gw

Hope this makes the issue clearer.

Best,
Klas

kissklas, 2015-09-10:
Good evening, sthalik! 

Finally got around to making a recording with Translation compensation on and off. In these videos, I only rotate my head, and move back and forth.

Here is the one with compensation enabled. As you see when the head is moving away from centered position, the rotation pivot does not move with the head position, resulting in wide rotations around the centered position:
https://youtu.be/V_u099NOj_Y

Here is the same movements without translation compensation. All rotations seem correct, no matter where I move the head (The only problem of course, is that I don't have compensation when looking backwards.):
https://youtu.be/VMTh9E6Y9Gw

Hope this makes the issue clearer.
Best,

MathijsG, 2015-09-13:
Same problem with Rise of Flight indeed.

kissklas, 2015-09-16:
Looked into this a bit more, and I found this is only a problem using the config setup for older Track-IR models in IL-2: BoS.  This file is found here: "<GAME FOLDER>/data/startup.cfg": Due to problems with the newer trackers during beta, I set the "old_trackir =" to "1". Reverted it to "0" now, and Translation Compensation works like a charm.

And @MathijsG... 
I looked in the Rise of Flight folder as well (being based on the same engine). Found the file but no "old_trackir =". They have however updated RoF since IL-2: BoS came out, so you could try just adding it, like so:

[KEY = input]
	input_map = "input.map"
	mouse_plane_control = 0
	old_trackir = 0

I guess you could close this thread @sthalik, since it turned out to be a game issue.

Best,
Klas

sthalik, 2015-09-16:
@kissklas thanks for the report. But if @MathijsG says Rise of Flight is broken too, we need to wait on this one.

sthalik, 2015-10-05:
Is it that translation compensation is already applied with "old_trackir = 1"?

kissklas, 2015-10-05:
I do not believe so. It was the lack of compensation that made me activate yours. If it is already applied from the games side with "old_trackir = 1", it does not seem to really work at all. I will double check this when I have time.

On a different note:
I did however notice a little bit of strange rotation this weekend while playing IL-2: BoS with "old_trackir = 0" and translation compensation on. There seemed to be a bit of offset, much like what I described with "old_trackir = 1". But on a very small scale. It seemed fine unless I looked over -90/+90 degrees in game and rotated my head vertically. Not sure if this is a result of bad calibration though.

I can do another video if I am able to replicate it after a calibration.

Best,
Klas

kissklas, 2015-10-05:
There is definitively no Translation Compensation implemented in the old trackIR setting.

I was not able to replicate the issue I was having either. It was probably just the calibration then.

sthalik, 2015-10-06:
@MathijsG are we fine?

kissklas, 2015-10-06:
@sthalik unless he replies, I can probably check ROF later, if "old_trackir = 0" works.

MathijsG, 2015-10-06:
I will check today. I didn't use the newest OpenTrack yet because it had some weird positioning, but I will try today.

Where do I need to put the old_trackir parameter? In settings of game or profile of Opentrack?

kissklas, 2015-10-06:
Check RoF folder for "startup.cfg". Under "[KEY = input]" add the sentence like so:

[KEY = input]
input_map = "input.map"
mouse_plane_control = 0
old_trackir = 0

MathijsG, 2015-10-06:
With that setting enabled I got no problems anymore. But it isn't very user friendly I think to let people alter settings? Why DOES it work without translation compensation and when you enable it (without editting the config) the rotation point is wrong.

kissklas, 2015-10-06:
It is a setting that is standard in IL-2: Battle of Stalingrad, which is build from the same engine. Probably added updated trackIR support since that game came out a lot later than RoF. They have since pushed some of the changes made in IL-2 back to RoF and updated the engine. I guess this "hidden setting" is one of them. 

You could notify the developers, and ask them to set the old_trackir = 0 as default in RoF, as it fixes translation compensation issues.

MathijsG, 2015-10-06:
I submitted a ticket to the Rise of Flight team. Will keep you guys updated.

sthalik, 2015-10-06:
Don't, it breaks freetrack with = 0.

MathijsG, 2015-10-06:
What do you mean? It is working perfectly now.

kissklas, 2015-10-06:
On opentrack, yes. 

I think what sthalik means is that freetrack, another popular free headtracking software (which is no longer being developed I think?), does not play well with the "new" setting (old_trackir = 0).

MathijsG, 2015-10-06:
But then the problem is in this software instead of in the game I think?

kissklas, 2015-10-06:
There is no problem with Opentrack. I would say it is Freetrack and older TrackIR versions that are getting outdated, and that is creating compatibility issues. People are still using older software, and the game does not recognise what you use automatically. if they made the setting a tickbox in the UI, instead of hiding it in the config, that would probably be better. I wouldn't think that should be accommodated in Opentrack.

sthalik, 2015-10-06:
@MathijsG freetrack's problem is in baseless legal threats against the software. Give them a break. Look up "natural point of view" in google.

Better for the game's authors to fix compensation than to default to obfuscated protocol. We support the obfuscated one because we have to, not because we want to.

MathijsG, 2015-10-06:
I emailed the developers of Rise of Flight and they replied a bit annoyed. I don't think they will change anything.

kissklas, 2015-10-06:
Oh dear. I did not know about any of this. Yes, better not go to tech support again, as certain people might be involved in this. The respect and gratitude I have for the work you do here just went up a few more notches. Keep it up.

sthalik, 2015-10-06:
Thanks guys. Like @kissklas said let's leave implementors/game authors alone, too much heat is bad.

Since it seems to be all fine I'm marking resolved closed.

closed 2015-10-06 by sthalik

MathijsG, 2015-10-08:
With RC19, which is working great though, I got this problem again. (I already editted the config of ROF to put in the line). Sometimes after centering it is gone. But it looks like something still is wrong.

sthalik, 2015-10-09:
@MathijsG what is gone? Is opentrack forgetting settings?

MathijsG, 2015-10-11:
I don't know, but I got the same rotation problem sometimes, while I already changed the config of RoF. Will investigate further when not ill.

MathijsG, 2015-10-27:
This still is wrong here, even after changing the config. After I center again it's okay again.

Specs: PT headtracking,

[image]

sthalik, 2015-10-28:
What's wrong? I don't have the game to check myself.

MathijsG, 2015-10-28:
The same rotation problem as shown in the earlier video. You somehow rotate
around a point instead of rotating self.

Groeten,

Mathijs Groothuis