opentrack/opentrack #234: Linux joystick tracker support

runningman84, 2015-09-05:
It would be cool if opentrack would have a linux joystick tracker input. There are some cool devices like EDtracker pro which could work with it.

sthalik, 2016-06-28:
I'll be using the Qt gamepad module for linux and osx joystick support. Windows is too complex sadly.

sthalik, 2016-07-21:
Can someone write an implementation based on Qt gamepad support? It's trivial compared to evdev.

sthalik, 2016-07-21:
And it's gonna work on OSX also.

sthalik, 2018-05-14:
The QtGamepad class is way bad, need to implement it natively.

DanielKinsman, 2018-08-28:
Perhaps using SDL2 which could work for all OS?
https://wiki.libsdl.org/CategoryGameController

CrazyKnightGR, 2019-05-17:
trying to add edtracker support for linux. This is a analog joy emulation using x,y,z axis.
I have write a simply c program to find edtracker device and read x,y,z, but yet I don't fully understand how opentrack api works. Can someone help?

main.c
----------
#include <fcntl.h>
#include <unistd.h>
#include <string.h>
#include <stdio.h>
#include <linux/joystick.h>

int main ()
{
    struct js_event jev;
    char device[15];
    char name[128];
    char *s = NULL;
    int fd;

    for (int i=0; i<20; i++) {
        sprintf(device, "/dev/input/js%d",i);
        fd = open(device, O_RDONLY);
        if (ioctl(fd, JSIOCGNAME(sizeof(name)), name) > 0) {
            s = strstr(name,"EDTracker");
            if (s != NULL) break;
        }
    }

    while (s !=NULL) {
        read(fd, &jev, sizeof (jev));
        if ((jev.type & ~JS_EVENT_INIT) == JS_EVENT_AXIS) {
            switch (jev.number) {
                case 0: { printf("X=%i\n", jev.value); break; }
                case 1: { printf("Y=%i\n", jev.value); break; }
                case 2: { printf("Z=%i\n", jev.value); break; }
             }
        }
    }
}

sthalik, 2019-05-18:
See the [test tracker](https://github.com/opentrack/opentrack/tree/unstable/tracker-test).

sthalik, 2019-05-18:
Note, I took the liberty to reformat the code in your comment.

CrazyKnightGR, 2019-05-19:
@sthalik This was very simply. But anyway thanks for reply, I figure out how API works and tracker-edtracker works, but need some tests & performance tuning. After that will upload.

CrazyKnightGR, 2019-05-20:
I have a small problem with:
`
    notifier = new QSocketNotifier( iFile, QSocketNotifier::Read, this);
    connect(notifier, &QSocketNotifier::activated, this, &tracker_edtracker::updateAxis);
`
so maybe I'll follow QThread model, because the following code is very slow ... 
[alpha](https://www101.zippyshare.com/v/zxaZA1gb/file.html) This is without serial support and dialog axes not working.

sthalik, 2019-05-20:
Don't use signals/slots for this. See [tracker/udp](https://github.com/opentrack/opentrack/blob/unstable/tracker-udp/ftnoir_tracker_udp.cpp#L26) as an example.

For settings usage I don't have a self-contained complete example but see [declaration](https://github.com/opentrack/opentrack/blob/unstable/tracker-pt/pt-api.hpp) and [usage](https://github.com/opentrack/opentrack/blob/unstable/tracker-pt/ftnoir_tracker_pt_dialog.cpp). The settings API is always thread-safe (but using manual `QObject::connect`, not always).

CrazyKnightGR, 2019-05-23:
I have not available time for more options(retrieve-save eeprom), maybe next weekend will do, but until then, here a working [beta version](https://www98.zippyshare.com/v/wAmbH0Md/file.html).

alamo888PL, 2019-10-06:
Hello, can you post new link with your beta?

CrazyKnightGR, 2019-11-06:
@alamo888PL Sorry, had long time to login. If you haven't already find it, look source files at [opentrack](https://github.com/CrazyKnightGR/opentrack/tree/unstable/tracker-edtracker) and a compiled version (ARCH) at [opentrack-tracker-edtracker.so](https://www115.zippyshare.com/v/AGFQbfWC/file.html)