opentrack/opentrack #235: Center position

petsild, 2015-09-06:
Probably the best build rc17P2 in the center position, but yet I feel as if after a long time will reduce the look of mine checking through the crossfire, the center is not effective helps only to again stop start.

sthalik, 2015-09-07:
I don't understand. Please explain.

petsild, 2015-09-07:
After some time 10min I don't have the opportunity to be centering the image of the hot keys properly on the center, effectively works only the new start-stop.
Head I try to still stick to the default location.
I can create a short video where it will be better explained.

Would it be possible to add a hot key on the function of the Start-Stop?

sthalik, 2015-09-07:
You mean hotkeys don't work anymore?

petsild, 2015-09-07:
The image of the octopus to stay slightly below even after pressing the centering.

sthalik, 2015-09-07:
Adjust filter settings to lower deadzone.

petsild, 2015-09-07:
I use this settings
http://img8.rajce.idnes.cz/d0801/4/4901/4901891_507dca2b7c43a1bacef8c830a77145ff/images/Accela.jpg

MathijsG, 2015-09-08:
You are looking straight forward when pressing center?

petsild, 2015-09-08:
After setting the deadzone to 0 with the centering behaves correctly thank you very much.

sthalik, 2015-09-09:
@MathijsG it's not necessary.

sthalik, 2015-09-09:
@petsild does the problem still exist? Are we done here?

petsild, 2015-09-09:
Hi,
now I have tried the rc18 and centering responds quickly and accurately :D really great work.

closed 2015-09-16 by sthalik