opentrack/opentrack #236: Virtual joystick issue on OSX

rcapener, 2015-09-07:
I've set everything up and the phone and my mac are bluetooth tethered. The tracking is working as the little octopus is moving, and all is well - except that Elite Dangerous does not seem to see it at all.

I'm on a macbook pro. I've tried each protocol available (Flightgear, UDP reciever, and Wine) nothing seems to work and freetrack protocol doesn't seem to available for mac, and doesn't show in options.

sthalik, 2015-09-08:
We're gonna implement it using foohid, a driver working on raw HID descriptor data. This allows for creating arbitrary joysticks.

See: https://github.com/unbit/foohid
HID: http://eleccelerator.com/tutorial-about-usb-hid-report-descriptors/

rcapener, 2015-09-08:
Excellent! :)

rcapener, 2015-09-09:
Great news! Thanks for looking into this :)

sthalik, 2015-10-03:
@rcapener I don't think many users will want to install an unsigned kext (kernel module). The solution looks worse than the problem.

sthalik, 2015-10-06:
@rcapener I'm sorry but the kext solution is just bad. Only people with hackintoshes would use that at all. Barring better solution I'm closing the issue.

rcapener, 2015-10-06:
Understood. I'm very grateful to you for trying though, so thank you!

e4z9, 2017-11-19:
unbit provides signed kernel extension binaries with the [latest release](https://github.com/unbit/foohid/releases) (from mid 2016), so this sounds like a viable option after all?

sthalik, 2017-11-19:
I don't have

sthalik, 2017-11-19:
I don't have access to an OSX machine right now. Please consider contributing code. Thank you for the new info.

e4z9, 2017-11-19:
I've got a rough implementation, I'll try to polish it up a bit the next days.

closed 2017-12-02 by sthalik