opentrack/opentrack #237: The basic limit position

petsild, 2015-09-12:
Hi, I apologize in advance for the weird post. Friend on the rc18 has arisen the problem of the extreme positions, the movement of the octopus dodging into other positions. Has crookedly attached clip on the headset. The previous version rc14-17 functioned well without redoing the clip.
Rather I will add pictures.
[image]
[image]
[image]

If there are leds on the clip positioned exactly there is no difference in the end positions between the rc17-18. Thank you for understanding and sorry for the post.

sthalik, 2015-09-16:
This problem is related to centering only, right?

petsild, 2015-09-16:
If looked left the picture went diagonally, look to the right of the good.

sthalik, 2015-09-16:
Note to self: we need two centering methods, current as relative and previous as absolute.

petsild, 2015-09-16:
The clip what was all the time diagonally and on the older OT the work, is now aligned exactly and there is no problem. The attached pictures are now history. Thank you very much for your support and I look forward to another OT. Pet

commit 851116d8 (2015-10-02): main: add relative and absolute center options

Closes #237
closed 2015-10-02 by sthalik