opentrack/opentrack #238: Unwanted bezier curve S-shape in mappings.

YellowOnion, 2015-09-18:
I'm trying create a little dead zone, but there's a tiny "hump" near zero position, this is not just a visual bug, but easily seen while yawing/pitching in game as well.

I've repositioned a point to exagerate the effect in this picture: http://i.imgur.com/JQZJF0H.png

But even without any point near zero, it creates a 1-2px size hump like displayed here:

[image]

MathijsG, 2015-09-19:
Maybe something to change the curve strength between points? Now you can't make things lineair if you need.

sthalik, 2015-09-19:
Add a fourth point, it interpolates better then.

sthalik, 2015-09-20:
Actually, the bump is where your first point is. The function always passes the points exactly. I see no bug here.

YellowOnion, 2015-09-20:
sthalik, I don't see it as a bug but I don't see this behaviour as intuitive and expected either, when x increases y should always increase, that's the expected behaviour for me.

for a comparison, Blender uses a selectable Linear, Bezier and Constant interpolation for its animation system, it also gives you access to the handles of the Bezier curves to further refine expected behaviour.

I've tried adding extra points but some way to control curve strength and adding hard corners would be nicer, actually I think having an alternative to this mapping UI where you just control 3 parameters deadzone, sensitivity curve would be a lot more intuitive and cover 99% of use cases.

sthalik, 2015-09-20:
Monotonicity of the spline is one thing. Another is the derivative at given place between control points. For now there's not enough motivation to implement more interpolation method than Catmull-Rom that we're already using. But I'm leaving the issue open.

sthalik, 2015-10-30:
I'm closing this until there are code contributions.

closed 2015-10-30 by sthalik