opentrack/opentrack #239: ht - facetrack completly broken with 2.3 rc18

JoeSchr, 2015-09-22:
Hi,

I tried to use ht - facetrack with PS3 Eye cl driver. Its completly broken in this build. Position jumps frantically all over the place.

It works way better with the builds on dropbox from @sthalik at this issue thread: https://github.com/opentrack/opentrack/issues/163 which says it's opentrack-2.3-rc12-p1-46-geefe42b
Also with Facetrackernoir FaceApi it's totally fine. 

So something broke in this build. If you need anything else, I'm ready to help (but mostly only have time on the weekends)

JoeSchr, 2015-09-22:
So I just test some more: Sadly it's true, even when working better in the rc12 build, ht -- facetracker compares very poorly to faceAPI v3.2.6. Is it possible to load this Tracker into opentrack? Because opentrack is better on all the other stuff....

sthalik, 2015-09-23:
Before 2.3 release it needs to get fixed, yeah. I'm thinking of way simpler code than the HT face tracker. For now it stays like this, I'm busy like hell for several more weeks.

JoeSchr, 2015-09-23:
Ok, thank you for answering!

I know to code myself, do you think, it would be possible to get the FaceAPI to work with opentrack? Because their stuff works really well. If yes, could you give me some pointers in which direction I should start looking, if I want to make it work as plugin of opentrack?

Thx!

sthalik, 2015-09-24:
FaceAPI used to work with opentrack. It was removed since it was closed-source a long time ago.

You can add faceapi subprocess to the build, that's the important part. The old code still lives in repository history.

sthalik, 2015-10-06:
@JoeSchr you can also see wiki "hacking opentrack" page for an overview of the project's source code. Also see CMakeLists.txt for our easy-to-use build system.

sthalik, 2015-10-27:
Closing in favor of #197

closed 2015-10-27 by sthalik