opentrack/opentrack #240: Help!

FrankieHartman, 2015-09-23:
sorry if I'm asking you, but the people I know don't know don't able to help me.
I'm in mind to run code of an editor (Visual Studio 2010 and Oculus SDK for Windows V0.6.0.1-beta) and I don't know what to do ...
please help. I want to learn...
and sorry for the trouble

Only a quick guide as : create a project and load files in its folder

FrankieHartman, 2015-09-23:
I've downloaded version 2.2

sthalik, 2015-09-24:
Run 2.3 from unstable branch. It should mostly work with Visual Studio 2015. Older version have incomplete C++11 support.

FrankieHartman, 2015-09-24:
a friend of mine is creating a project with visual studio 2010. I have added to this project and I imposed this constraint ... then I repeat my question ... can someone tell me how to run this project on visual studio 2010? if I can get it started, with debugging, I can see everything I need

sthalik, 2015-09-25:
You can't. Period. Visual Studio has ancient C++03 support. It won't ever work.

closed 2015-09-25 by sthalik