opentrack/opentrack #242: "Zero" is not alligned perfectly to forward

nobkins, 2015-09-27:
I use OpenTrack with Elite Dangerous and an EDTracker. I also use screen capture to grab price information from within the game. For the screen capture to work you must be looking dead ahead or the capture data is unreliable due to screen screw on the OCR.

In OpenTrack (2.3 rc18) I have bound a key to the "Zero" option. The problem is that this does not return the screen to zero. In fact it seems as if it returns it to close to zero but the deviation is affected by where you where looking before you activated zero. The deviations is small but enough to cause problems for me.

I can tell there is a deviation because if I activate the Zero option by pressing the bound key and then click the stop button the camera view will move slightly when I press stop. If the zero was true zero the camera would not move.

I can also turn off the head look in game (but only if I use the vjoy option which in game is much more laggy compared to freetrack option) and this shows the same problem.

Here is a YouTube video to demonstrate the issue.
https://youtu.be/IIkl_FIKbew

Thanks for looking.

sthalik, 2015-09-27:
The filter ignores small movements, in this case leading to zero. Does it ever occur with filtering off?

nobkins, 2015-09-27:
I was using the Acela filter and with it turned off the same issue still occurs. If I zero the tracker and the press the stop button the game jumps to true zero only after the stop button is pressed.

sthalik, 2015-09-28:
Note to self: ignore filter when zero is held, pass zeros directly to game and interface.

nobkins, 2015-09-28:
Ok. My understanding of how all this is handled by the software is minimal but just to check....

From my previous post it would seem to indicate that the filter is not the cause of the problem. When I had exactly the same issue with no filter and with the Acela filter. The filter made no difference.

sthalik, 2015-09-28:
Filtering and centering both, to be precise. Both need to be ignored and ideal zeros output.

What game requires zeros? Is it Elite Dangerous menus?

nobkins, 2015-09-28:
The problem only causes me an issue in Elite Dangerous. And the issue is only because in Elite Dangerous I take screenshots of the space station commodity prices and then an OCR program reads them in so that I can quickly look up prices from my own database of stations I have visited (hope that makes sense).

Because of the lack of perfect zero the OCR's regularly don't work because it has to have everything lined up exactly the same way each time a screenshot is taken.

closed 2015-10-02 by sthalik

sthalik, 2015-10-08:
@nobkins rc19 released and includes the fix.

nobkins, 2015-10-08:
Superb I will give it a try and confirm the results.

nobkins, 2015-10-08:
Works perfectly now. Thank you very much.

commit 4117f4c6 (2015-10-02): main: keybinding for "zero" should reset to PERFECTLY forward

Closes #242