opentrack/opentrack #245: [bug] Cancelling action of choosing new executable resets current executable

MathijsG, 2015-10-03:
I have some games set up to automate the tracking:

[image]

When I click '...'  to search the executable (I expected to be directed to the current executable folder) which isn't.

The currently configured excutable gets zero'd.

sthalik, 2015-10-05:
@MathijsG and thanks for the effort of _all_ of the reports you've made.

MathijsG, 2015-10-05:
And is it also easy and possible to start at the location of the current set executable when you press '...'? Now you always start at a default location.

Thanks for making this software :)

sthalik, 2015-10-05:
ok fine wait some 😄

sthalik, 2015-10-05:
I can has emoji? 😈

MathijsG, 2015-10-05:
No :-P but I do see them in my notification e-mail. Weird Github.

sthalik, 2015-10-05:
strange, works for me on desktop site. but rdp doesn't so I can bring opentrack here. such sad.

sthalik, 2015-10-05:
@MathijsG we can't jump to current file's location because it's not stored, only the file's name is stored. It has to be that way sadly.

closed 2015-10-05 by sthalik

sthalik, 2015-10-06:
I got VNC set up. Expect hackage.

dbaarda, 2015-10-06:
Hmm... just read this after talking to someone about my Cygwin/VNC/NX/X2GO
painful experience trying to get qtcreator to work remotely. I documented
my woes here;

http://minkirri.apana.org.au/wiki/DevJournal

It might help understand things if you encounter problems like I did...

sthalik, 2015-10-06:
Why this convoluted experience? I run Qt Creator with Cygwin cross-compiler kit using stock Cygwin-provided mingw-w64 compilers. There are some pathname discrepancies but nothing important to the build process.

dbaarda, 2015-10-06:
At the time I wanted to run qtcreator on the linux box to have a pure Linux
devel/build environment. My only Linux box was a headless server, and the
workstation with screen/keyboard was my gaming box running windows. I
wanted to avoid dual boot or linux-in-a-VM-on-windows stuff.

What amazed me was how little of what I tried worked. You would think VNC
client or Cygwin x-server on windows box doing remote-desktop access to a
Linux server would be a no-brainer, but I kept finding new ways that it
wouldn't work.

sthalik, 2015-10-11:
@dbaarda see https://github.com/opentrack/opentrack/wiki/building-on-cygwin-targetting-windows

commit b71978bf (2015-10-03): process-detector: check when cancel button is pressed

Closes #245