opentrack/opentrack #246: Is there a magic trick to get OpenTrack to work in iRacing?

jasondilworth56, 2015-10-05:
I hope someone can help. I installed OpenTrack today, and am using an IR clip and PS3 Eye. OpenTrack is registering perfectly well. I leave that running, open iracing.com and start a session from there. I then go to options to assign the 'look' commands, but the IR setup is seemingly not recognised. Is there something I'm doing wrong?

ph0b, 2015-10-05:
you shouldn't have to do any configuration inside iRacing to get it to work.
Select the freetrack protocol and start opentrack, tracking will work while you're driving.

sthalik, 2015-10-06:
I second that. If there are any problems with iracing and freetrack protocol, it needs investingating though.

sthalik, 2015-10-06:
If it's still broken, try moving to "program files" or "c:/". Might be a filesystem permission issue.

jasondilworth56, 2015-10-06:
Thanks for the replies guys. It turned out to be what I would call a bit of a bug. Within iRacing I had 'Look Left' and 'Look Right' bound to the left and right keys on my wheel. All the time that was the case, I couldn't use OpenTrack and I couldn't bind those actions to the yaw axis. 

To fix the problem, I had to bind both 'Look Left' and 'Look Right' to Space, which set them both to Unassigned instead. Once they were unassigned the tracking started working instantly.

sthalik, 2015-10-06:
glad you sorted that out

closed 2015-10-06 by sthalik

bodylinksystems, 2015-11-10:
this gives me hope for using this with ETS2

sniperwolfpk5, 2015-11-11:
Well I played ETS2 with factracknoir. I think it should work with OpenTrack

bodylinksystems, 2015-11-11:
I'll give it another shot