opentrack/opentrack #247: elite dangerous completly broken with 2.3 rc18 after patch 1.4

brian19876, 2015-10-06:
elite dangerous completely broken with 2.3 rc18 after patch 1.4 no head traking at all

sthalik, 2015-10-06:
I'm sorry but can someone else confirm this too?

brian19876, 2015-10-06:
https://forums.frontier.co.uk/showthread.php?t=188414 here is a post from elite fourm

sthalik, 2015-10-07:
Holy jumping jethros, that's big. Replace this dll file in opentrack directory https://www.dropbox.com/s/71r8xx3jd6ouzek/NPClient-debug-20151007.7z

Run ED then send me <code>c:/npclient.log</code>

brian19876, 2015-10-07:
i replaced the dll file but searched my pc did not find any npclient.log file maybe i did something wrong i started open track then started elite for a few min and closed it

sthalik, 2015-10-07:
It gets created on top of the C hard drive. Run the game process as admin, maybe it'll get it created.

brian19876, 2015-10-07:
I did look at the main c drive I even searched the whole PC for it. I did try to run the game and open track both as admin still no file created i will try looking more into it when i get home from work Remote desktop does not work very well with full screen games lol

brian19876, 2015-10-07:
i moved opentrack to my c drive and clicked the option to locate dll and picked the 1 you sent me and for some reason the tracking is now working in the game if i run it from my drive G i get no tracking don't know why moving it the c drive made it work I posted on the elite fourms to see if anyone else gets it to work by moving it. By the way still no log file

sthalik, 2015-10-08:
Does it work when moving with the old dll though?

sthalik, 2015-10-08:
Note to self: if new DLL works with ED, we can add an option to ft protocol to choose between them.

brian19876, 2015-10-08:
ok i putt the old dll back in and it seems to work but only if the software is run from the C: drive moving the install to the c drive has also worked for other people on the elite dangerous forums that use opentrack. This does not make much sense to me but i'm no expert. I did notice that pitch and roll need inverted on old DLL but don't on the new one not a big deal im guessing this was a fix you made. If you need me to test or anything else out let me know and thanks for working on this great software i enjoy using everyday

sthalik, 2015-10-08:
It's directory permissions. The game process doesn't seem to be permitted to access the freetrack .dll. That is outside opentrack's control.

sthalik, 2015-10-08:
@brian19876 the 2.3rc19 release now has an installer file. Please check if installing to the default location solves the issue. I'll make an announcement on Frontier forum if it does.

brian19876, 2015-10-08:
ok I will try this in a few hours when i get home from work

brian19876, 2015-10-08:
i used your install seems to work fine using default install location thanks for all your work

closed 2015-10-09 by sthalik