opentrack/opentrack #248: Shaky View

Denco96, 2015-10-07:
Hi,

I get a very shaky view specifiacly when I turn to the left with the 2.3 r18 version. I had the same shaky view when zooming in the 2.3 r12p1. Did a quick test with FaceTrackNoIR and my view does not shake at all. Currently it is unusable for me using DelanClip with PS3 Eye.

I'm using the point tracker with accela and FreeTrack Protocol.

Regards

sthalik, 2015-10-07:
Hey,

The issue is known and in the process of getting fixed. What you can do is try this test build https://www.dropbox.com/s/yukqw3gwht475og/opentrack-2.3-rc18-39-g471c6ef.zip and confirm if it's still broken or not.

-sh

Denco96, 2015-10-07:
Will do mate. I will report back as soon as I try it. 

Thx :+1:

MarcinSzczecin, 2015-10-07:
Witam:) przepraszam że po polsku( mam nadzieje że można). Również potwierdzam mocne drgania zwłaszcza jak się patrzy na lewą stronę w rc_18 jak i w tych plikach z dropboxu. W rc_15 wszystko gra. Zamieszczam 3 linki gdzie to uwieczniłem.
Jest jeszcze jeden problem we wszystkich obu wersjach gdy kamera przechodzi przez centrum ekranu następuje mocne szarpnięcie, widać to bardzo dobrze na filmie z rc_15.

Stuttering on the left side and jerk in the center on Rc_18
https://youtu.be/_GByv1JOBBM

Only Jerk in the center
https://youtu.be/a-Th9V1Z-tY

settings
https://youtu.be/c7WmWobwqPY

A tak ogólnie to duże podziękowania za to że poświęcasz się temu projektowi. Opentrack to naprawdę kawał dobrej roboty :)

Denco96, 2015-10-07:
The link you posted seems to fix the problem but there is another one still present. At the center there seems to be deadzone present which jerks the camera. Please note that I don't have any deadzone set inside accela or mapping.

sthalik, 2015-10-07:
The center jerking was introducted shortly after rc18. I think it now doesn't do either that, or the previous Accela jumpiness. See https://www.dropbox.com/s/cnte080zrx83120/opentrack-2.3-rc18-42-g5441b2a.zip

Please confirm. If this works, we should be good now.

Denco96, 2015-10-07:
I will test it first thing tomorrow after I get home from work.

sthalik, 2015-10-08:
I think we're good, two people said it's fine. I'm closing this issue but if there's still something you can respond/reopen.

@Denco96 @MarcinSzczecin It's fixed in rc19.

closed 2015-10-08 by sthalik