opentrack/opentrack #252: Hot key to Start/Stop

petsild, 2015-10-10:
Hi, great work, I commend the last rc Opentrack are great, the control works absolutely excellently in simulators such as Rise of Flight and DCS World. I asked if it was possible in the future to select the hot key to Start/stop completely independently on the choice of the set of profiles?
Thank you for the perfect product hi. Pet

sthalik, 2015-10-10:
There's "game detection" in options, isn't that enough?

sthalik, 2015-10-10:
Technically we could do that shortcut now, following recent changes shortcuts outside of tracking work. But what for?

petsild, 2015-10-10:
Functions suitable to be a longer pause camera stop, without having to switch between windows and the crease so the mouse on the main screen of the OT.

MathijsG, 2015-10-20:
Isn't this suitable to use a key for zero? Then you have the same as not tracking?

sthalik, 2015-10-20:
I understand that tracking uses resources, and blocks the camera while in progress.

Sovereign-Man, 2015-11-11:
I would welcome a start/stop key, OpenTrack stops working quite often in Elite:D.
It doesn't crash per se, it either stops reporting raw data or stops sending that data to Elite.
I'm using rc19p2 and an EDTracker2

sthalik, 2015-11-11:
There was a tracking data freeze bug fixed in rc19p3

Sovereign-Man, 2015-11-11:
That was a quick reply :) will try p3 now.

sthalik, 2015-11-11:
How many of you guys want it, other than people already in the thread? Drop a "+1" or like that.

petsild, 2015-11-11:
The other two friends of Leon and Joska thank you.

severgun, 2015-11-26:
+1 
Why not? I think that this feature just copy and paste of code from zeroing hotkey to start/stop.
Isn't it?

sthalik, 2015-12-05:
@severgun no, we have different subsystem listening for stuff during tracking. That subsystem doesn't exist when tracking's stopped. See latest changes.

Who's feeling adventurous? Build - https://db.tt/vVs03dfD

petsild, 2015-12-06:
Such a function there was missing, great work thank you very much hello. Pet

sthalik, 2015-12-06:
We're done with the functionality. Report regressions in #267.

closed 2015-12-06 by sthalik