opentrack/opentrack #253: [bug] Centering sometimes doesn't respond

MathijsG, 2015-10-10:
While this new version is as fluid as never before, everything feels so smooth even though some games I play don't even produce a lot of FPS. Thanks :)

However I do encounter something like a not responding center button. I mapped the center to 'Home' and the Toggle button to 'End'. Sometimes I need to press Home multiple times, or even need to press 'End' first, and after that Home is working.

Other people encountering this?

Len62, 2015-10-10:
I have center set to shift+c and I have to press it twice for it to take. Haven't checked other buttons yet.

sthalik, 2015-10-11:
I'm having same issue. Will do.

sthalik, 2015-10-11:
Confirm the build please https://www.dropbox.com/s/mc2s5ke20kc1h9p/opentrack-2.3-rc19-5-g9ba347d.zip and we'll release rc19p1

closed 2015-10-11 by sthalik

Len62, 2015-10-11:
Centering working fine with this build. Thank you.