opentrack/opentrack #254: Won't center in Arma 3

tnbranko, 2015-10-11:
I've had OpenTrack working for a long time in Arma 3, but after being gone for a few months and coming back to it recently, I now have the problem of not being able to center or zero my view while in Arma 3. It works fine when I minimize Arma 3 and in every other game.

Has anyone else had this problem or know of any possible solution I could try? I just tried FRAPS and it enables when I'm in Arma 3, so it's kind of a strange issue, where I'm not sure if the problem is Arma 3 or OpenTrack.

sthalik, 2015-10-11:
Don't run the game or opentrack as admin

tnbranko, 2015-10-11:
Thank you so much! I had to disable "run as admin" in the properties for Arma 3 and now it works perfectly.

closed 2015-10-11 by tnbranko