opentrack/opentrack #255: DCS A-10C problem with 2.3 rc19 !!!!

Shadoware, 2015-10-13:
I was using rc 17 which works very good for me. I tried rc 19 and it seems more smooth but in dcs a-10, suddenly it stops! The soft doesnt crash but the game stops receiving information from opentrack and I cant change the view. If I use rc 17 I dont have any problem.

sthalik, 2015-10-13:
Can you reproduce? How long does it take for it to break?

sthalik, 2015-10-13:
Do you use headtracker.dll?

Len62, 2015-10-13:
I experienced this once myself. OpenTrack just stops, doesn't crash, still reading points but no output. I'll check again to see if it does it again.

sthalik, 2015-10-13:
opentrack has no output shown in the UI or just doesn't pass to game?

Do you have headtracker.dll or use freetrack as-is?

Len62, 2015-10-13:
No output in the UI. I can click stop, then start, and it starts again. Using freetrack. I just ran it again for about 30 minutes w/o any issues.  Will do some more testing and report back.

MathijsG, 2015-10-13:
I maybe encountered the same in ArmA II. Will play, ehh investigate further.

Len62, 2015-10-13:
Just had the issue again after about 45 minutes of running. If I switch to the UI, the points are tracking, the raw data is displaying properly, but game data is stuck on 0.

petsild, 2015-10-13:
I have the same problem stopping the data the rc19p1. So far did this to me randomly around 30min Rise of Flight and DCS, will only help to put the stop and run again. I use Win7 home premium 64-bit and the games are non steam. I'm trying to still rc18 defect has so far not exhibited.

Len62, 2015-10-13:
One more time after around 40 minutes in BOS. Notice the game data values in the pic. Interesting since I have TY and Roll disabled.
[image] Points still track and raw data working. Just the game data locking up.

sthalik, 2015-10-13:
try press toggle keybinding. it maybe can trigger on its own so bind it and report back.

petsild, 2015-10-13:
I will add too the picture just after the failure time 35 min after power up.
[image]

sthalik, 2015-10-13:
each of you please tell if you have zero and our toggle bound

Len62, 2015-10-13:
Yes, I have both bound. After it locks-up toggle does not have any effect, zero will re-set game values to zero.

petsild, 2015-10-13:
Exactly as he wrote Len62.

Len62, 2015-10-14:
Update, ran opentrack for a couple of hours this evening w/o any problems, no lock-ups, no glitches. Will test some more in the morning.

sthalik, 2015-10-14:
@Len62 could it be that you haven't had toggle/center bound but now you have?

sthalik, 2015-10-14:
there's no check whether keybinding is disabled. maybe that's the cause. control space or some other key combination could generate an ASCII zero.

Len62, 2015-10-14:
No, all 3 of the bindings were set. I'll have another go with it shortly and see what happens.

Len62, 2015-10-14:
Ran another session around 60 minutes w/o any issues. Strange to say the least.

sthalik, 2015-10-14:
this can also be filter-related. I need to install Windows again, stay tuned.

Len62, 2015-10-14:
Had a lock-up after about 30 minutes just a short while ago. Auto-threshold maybe ? I really didn't check to see when I had it enabled. I'll do some more checking and report back.

Len62, 2015-10-16:
Auto-threshold does not appear to be connected to this issue. The lock-up is still intermittent and I can't really get any idea what causes it. Last night experienced a couple of lock-ups, but this morning none while running for about 60 minutes.

sthalik, 2015-10-17:
Try this one https://www.dropbox.com/s/pwg1srxazw6ydft/opentrack-2.3-rc19p1-9-gf314ef0.zip it has two speculative fixes.

If it works for a few days for you guys I'll release p2.

petsild, 2015-10-17:
Hi, I already wanted to write that after an hour of testing is in order, but then chobotnička again stopped. But it may now be repaired by the center and the movement again works. Yesterday I used mp on rc19 and everything worked well without stopping the 3h in a row. Friends make the image even once, did not stop for the entire time of use rc19p1, I have no idea what's causing it. Much I appreciate your efforts and perfect support thank you. Pet

sthalik, 2015-10-17:
so it's the filter 99%.

sthalik, 2015-10-17:
Good morning Vice City!

https://db.tt/Py90BwBL

MathijsG, 2015-10-17:
I had it once yesterday in Rise of Flight. After restarting it and flying for at least the same length, no problems then. I have auto treshold set to off.

sthalik, 2015-10-17:
@MathijsG try the today's build. I think I know where the problem resides, now.

Len62, 2015-10-17:
Ran the latest build approx. 90 minutes w/o any issues.

sthalik, 2015-10-17:
@Len62 how long was the loss?

Does it lock up completely with no recourse other than recenter still?

sthalik, 2015-10-17:
@Len62 are your model coordinates accurate? It's likely that PT submits broken coordinates for some reason.

Len62, 2015-10-17:
The loss is only momentary, a second or less most times, like a blink. No recent lock-ups, just the "blink". I'll double check the model coordinates and get back with you.

sthalik, 2015-10-17:
@Len62 does the loss merely drop frames or does it reset to another pose?

Len62, 2015-10-17:
Just drops frames. I went back and checked my set-up, model coordinates were fine. I did change the camera pitch angle, as I recently raised my monitor up some, and re-did the calibration. Seems to be working fine right now, haven't had any issues since I re-checked all of my settings.

petsild, 2015-10-17:
I tested the latest build but still the same problem. The last time happened in a completely direct view.

sthalik, 2015-10-17:
@petsild do you have opentrack-2.3-rc19p1-23 in the title bar?

petsild, 2015-10-17:
Yes rc19p1-23

sthalik, 2015-10-17:
And you need to center then? I find that hard to see now that each code path in tracker pipeline checks for invalid values.

petsild, 2015-10-17:
After centering the turn continues normally.

Oakshios, 2015-10-18:
I have had this issue using Freepie with Freetrack enhanced protocol.  Will try with the newest build and report back.

sthalik, 2015-10-18:
@Oakshios unpack the .zip anywhere you want and run it: https://db.tt/Py90BwBL

It's important knowing that FreePIE gives the same result. It's definitely mapping- and filter-related.

sthalik, 2015-10-18:
New build with more checks in Accela and mapping editor (involves Accela too) - https://db.tt/D5luNvTe

Oakshios, 2015-10-18:
Thanks for the zip.  Installed and will let you know if I notice further freezes.  Let me know if there is any other config info you might need to help narrow down the issue.

sthalik, 2015-10-18:
Just make sure model coordinates and camera FOV are ok.

Oakshios, 2015-10-18:
Using rc19-30 I am still getting freezing.  When it happens sometimes the coords lock in some random crazy position.

sthalik, 2015-10-18:
Are you using dynamic pose resolution? Here's a fix relating to dynamic pose and this problem. https://db.tt/ixlixsbN

Oakshios, 2015-10-18:
I am not sure what dynamic pose is.

sthalik, 2015-10-18:
PT settings -> camera -> dynamic pose resolution

Oakshios, 2015-10-18:
I found that setting, but I don't use that tracker.  I use freepie imu on my Galaxy Note 4 and the freepie UDP tracker to accept the raw data.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Oakshios, 2015-10-18:
I am calling it a night on my end.  Will check back later today.

sthalik, 2015-10-19:
I need everyone's input on https://db.tt/ixlixsbN, especially on PT tracker.

Len62, 2015-10-19:
I still get the occasional tracking loss,  and I did get a hard lock-up once last night, and stop did not work, clicking the X to close the window caused it to crash completely. I'll be doing some more testing over the next hour or so.

sthalik, 2015-10-19:
note to self: check valgrind

Len62, 2015-10-19:
Here's a short video showing what is happening. 
https://www.dropbox.com/s/8cpcpavla92gsp9/opentrack.mp4?dl=0

Len62, 2015-10-19:
I deleted the previous post and I have to apologize to you. I did some checking on my clip and discovered that one of the power wires had a connection that was failing intermittently, enough to cause the LEDs to turn-off sporadically when I pitched down. Should have thought to check this in the first place. I soldered up a new connection and everything works fine. Sorry.

petsild, 2015-10-19:
My cable is fine, after testing 3x2h and efforts to simulate by the conditions of the freeze is build 36 stable.

Oakshios, 2015-10-19:
I will confirm stability with new build on my configuration when I get home from work.  Thanks for hunting down this issue sthalik!

sthalik, 2015-10-20:
@Len62 do I understand in the video (thanks for not removing it just yet) it fails interminnently but never forever until recentering?

@petsild do you mean that it works or it doesn't?

sthalik, 2015-10-20:
When works for everyone I'll update patchlevel.

sthalik, 2015-10-20:
@Len62
> I deleted the previous post and I have to apologize to you. I did some checking on my clip and 
> discovered that one of the power wires had a connection that was failing intermittently, enough to 
> cause the LEDs to turn-off sporadically when I pitched down.

Your interminnent failure helped cause the error to occur, which is good. PT shouldn't break forever like it does in the current official build.

[44 more replies]

closed 2015-10-28 by sthalik