opentrack/opentrack #257: right-hand Ctrl, Alt, and Shift keys cannot be bound. v2.3-rc19

SgtGrumbles, 2015-10-17:
When binding Global Shortcuts for Center, Toggle, or Zero, I cannot use the Control, Alt, or Shift keys on the right-hand side of the keyboard.  The left side versions work.

e.g. if I try to bind for Center, and hit (right) Ctrl + P, it'll be set to just "P".  Similarly, if I use the left Control + P, the bind will correctly read "Ctrl+P", but only the left Control + P will Center; hitting right Control + P will do nothing.

This is on a (fairly standard) keyboard with a US layout.

closed 2015-10-17 by sthalik

SgtGrumbles, 2015-10-17:
Holy schnikes that was quick!  Thanks.

sthalik, 2015-10-17:
@SgtGrumbles use https://www.dropbox.com/s/ptjxuqo5l6xfyvw/opentrack-2.3-rc19p1-12-g86418fa.zip?dl=0

SgtGrumbles, 2015-10-17:
Tested and verified working.  Thanks again!

commit c47260fb (2015-10-17): shortcuts: alias right modifier keys to left modifier keys

Issue: #257
Closes #257