opentrack/opentrack #258: Center and toggle key binds not working in 2.3RC19p2

DerangedPineapple, 2015-10-24:
Center and toggle key binds won't work at all if they both are bound to same key i.e. I had both Center and Toggle bound to Return (keyboard Enter) in 2.3RC18 and after updating to 2.3RC19p2 those key binds wont work anymore when bound to same key (doesn't matter which key).

If I bind them to different keys they work just fine as well as Zero works fine with Center or Toggle when bound to same key.

DerangedPineapple, 2015-10-24:
Update as I Forgot to mention OS and tracker in use.

I have Windows 10 upgraded from Win7 and tracking device is EDTracker configured in opentrack as joystick input and outputting it TrackIR and Freetrack and running opentrack as administrator.

There is small difference how Zero works comparing to Center and Toggle functioning together. I would rather use Center+Toggle combination as after releasing Toggle tracker would be centered also, with Zero opentrack keeps tracking head movement and this can lead in my case to funny situations.

sthalik, 2015-10-24:
This is an unusual use case but there you have it now.

closed 2015-10-24 by sthalik

DerangedPineapple, 2015-10-24:
Thank you. It's very useful feature when you need to aim down to sight in DayZ/Arma2-3 and center  and toggle your head tracking at the same time so the view won't drift accidentally when turning head.

commit 603130ef (2015-10-24): shortcuts: allow for binding same key to multiple functions

Closes #258