opentrack/opentrack #259: Can't bind the center command to my numeric pad...

Hardcard, 2015-10-25:
Hi all.
I'm using Opentrack 2.3 rc19p1 and today I noticed that I can't bind the "center" command to any of my numeric pad keys...This is weird and unfortunate, I'm used to bind the "center" command to the "5" key of the numeric pad, it has always worked in Facetracknoir and Freetrack, but it doesn't seem to work in Opentrack, sadly. 
I'm guessing there is some kind of driver incompatibility...
Am I the only one having this issue?

sthalik, 2015-10-25:
Qt has no constants for numpad numbers. I can alias them to normal numbers at best.

sthalik, 2015-10-25:
Note: they aren't aliases, use a "Num" modifier that works similar to shift or ctrl.

commit 18efc635 (2015-10-25): shortcuts: allow for numpad on win32

Closes #259
closed 2015-10-25 by sthalik