opentrack/opentrack #260: There is roll applied with pitch/yaw FT + PT+ Accela.  Using a 3 point tracker .

HellfireLx, 2015-10-25:
Depending on centering method, it is applied to yaw or pitch, never both.

Centering method:
- Absolute: Roll is applied on pitch
- Relative: Roll is applied on yaw

sthalik, 2015-10-26:
Check if your model is accurate. Works for me.

sthalik, 2015-10-26:
The "absolute" centering methods depends on model to be centered with regard to camera - you don't want it, it's suitable for inertial trackers only. You want the "relative" method instead.

sthalik, 2015-10-26:
Also see "camera offset" in options, it affects centering in a big way.

HellfireLx, 2015-10-26:
I think the model is not far from accurate, but I will measure/adjust , just to be sure. It might be that centering offset is the problem, I never applied any, the roll effect was never as strong as now (I used r15 until now, as it gave me the best experience and I play in team 3 times a week, so I can't experiment around a lot).

I am sorry to bother you about that, but when my model, which is the 3 point vertical one (like trackIR), is fixed on the right of my headset, what point on the headset should I take as 'center' ? I have to apply negative x axis, that seems logical , but for y axis, I wonder where I consider the middle. The middle LED in the center, is that the way to go?
Dziękuję bardzo !

sthalik, 2015-10-26:
See https://github.com/opentrack/opentrack/wiki/choosing-camera-offset it's gonna be linked from next version.

See https://github.com/opentrack/opentrack/wiki/model-calibration-for-PT-and-Aruco-trackers for centering.

Wporzo!

HellfireLx, 2015-10-27:
ok, I got it now. I had to read up the official TrackIR quick manual (I only have a self made clip) to understand that the best is to decenter the camera as well, so it is aligned with the model. That gets me much better results, I only had to adjust the pitch, which made totally sense.

Thanks again and keep up the good work !

HellfireLx, 2015-11-16:
Hi , 

I re-open this, in fact the symptoms are still as described in the opening post. I thought it was gone , but it is still there , now I disabled all axis output , but roll is still applied to it. 
pic of config : http://imgur.com/nOZTY4C

sthalik, 2015-11-17:
is it bad on both centering methods?

HellfireLx, 2015-11-18:
Sorry for this late reply.
Yes it has the same amplitude on both centering methods. Just that the axis which gets influenced by the roll (that is probably because I don't get the centering 100% right for roll) , is pitch for absolute and yaw fro relative. 
I know I have a problem with roll and centering, that's why I don't use roll (I use translation axis X , but for the repro example I disabled that too) . But when disabled as an input -> output , I would expect it to not be interfering at all, means , the roll game output value, should stay at 0 . It is as if the roll axis -> disabled choice was ignored when calculating roll game output.

HellfireLx, 2015-11-19:
FYI, as a workaround to eliminate roll on output, I went into the mappings and flattened roll curve to 0. Now roll axis is not output anymore.

sthalik, 2015-11-21:
it's not a bug. centering can and will add roll output despite zero raw value. disabling roll mapping only removes the raw roll amount.

HellfireLx, 2015-11-21:
Ok fair enough, I found a way to disable roll influence for me, as stated above. Unfortunately,I just seem  to be unable to get the centering right with my setup. Thanks.

closed 2015-11-21 by HellfireLx