opentrack/opentrack #261: Head tracking not working in Elite: Dangerous

NaturalHate, 2015-10-26:
In response to: https://forums.frontier.co.uk/showthread.php?t=188414&p=2929212&viewfull=1#post2929212  

First of all sorry for taking so long but my life kinda took a hit and kept me too busy to bother with it.

The problem still persists by the looks of it. If it's still a permission issue then I'm not entirely sure what to do or how to find that out exactly.

sthalik, 2015-10-26:
Did you run the installer version?

NaturalHate, 2015-10-26:
Yes, even though I tried moving the OpenTrack folder to C: before that too.
Again: It is just in ED not working, I tested Evochron and the Euro Truck Sim demo and in both of those games the head tracking works.

sthalik, 2015-10-27:
Folder properties -> Security -> Advanced -> Modify -> add yourself permissions for folders, subfolders and files for opentrack. Check "replace permissions". Ensure "Everyone" has read/traverse directory permissions for parent directories of opentrack.

sthalik, 2015-10-27:
Also install to the same drive as ED.

NaturalHate, 2015-10-27:
Seems to be correct, I've still tried to apply it for subfolders & files again and did that with my ED folder too but that didn't worked either.

They were both on the same partition already (except for the "installed" newer one on C:) but I moved ED to a different partition and also tried moving both versions of OpenTrack one after each other without any luck.

sthalik, 2015-10-27:
I'd like to get to the bottom of this, as plenty users are affected. Add me on skype "sthalik", I'll run some diagnostics.

NaturalHate, 2015-10-27:
I have to install that later because I don't use Skype and might have to do some stuff today.

I think most other users have this since 1.4 though - mine broke in 1.3 during my tests with Windows 10 but somehow stayed broken even after a fresh Windows 7 installation. Like I said it was working fine in other games, I suspected the TrackIR dummy to weird out at first but I think I've tested it by only using that in the other titles and it worked.

Is there a way to read out EDs communication attempts with TrackIR somehow?

sthalik, 2015-10-27:
maybe we can ask mbrookes

NaturalHate, 2015-10-27:
I had a very lengthy discussion with CMDR Vanguard via support ticket and they tried to "break" TrackIR internally to reproduce my issue. They couldn't though and closed the ticket at some point suggesting to use DependencyWalker to see if I miss some dependencies.

Couldn't find anything though and I also couldn't find anything via Google but tried to install all the redistributables from MS for that matter without any luck.

sthalik, 2015-10-27:
dependency walker is fine as opentrack is self-contained. I'll ask Brookes.

sthalik, 2015-11-01:
He's not responding, fuck it. I need access to a machine where this issue happens and to install some debug software.

NaturalHate, 2015-11-01:
Well, I can't just give you full access to my PC (which is neither Polish or English anyway) but I'll try to help if I can.

I also doubt they're allowed to talk too much about it, the TrackIR devs aren't the best as we already know ...

Kinda got the axis binding from edtracker to Elite working by the way, I don't know why that worked this time but Elite kinda resets some of those settings every freaking time I open up the controls menu which is fairly annoying. And I still want the proper head tracking + filters + tilting. The VJoy output wasn't working back then though, and I had to remove the VJoy drivers too since they cause problems with Unity based games but I guess we're back at OpenTracks and Elites communication for that matter again.

sthalik, 2015-11-01:
Can you run sysinternals procmon and see if npclient.dll is opened at all?

NaturalHate, 2015-11-01:
It shows it under Detail for the opentrack.exe at least?
Is that correct?

sthalik, 2015-11-01:
opentrack never opens npclient.dll. The game should at least try to load it.

sthalik, 2015-11-01:
Oh, and you can have a VM with ED installed in it.

NaturalHate, 2015-11-01:
I filtered the ED.exe (Including EliteDangerous32.exe) and searched (CTRL+F) for the npclient.dll without finding it, I guess that's a no then?

And how's a virtual machine representing my machine being the one with the problem though?

NaturalHate, 2015-11-01:
There you go ... 
Sorry for the derp.
http://i.imgur.com/BdmfRsA.png

NaturalHate, 2015-11-01:
Errr ...
It works now? 
What happened?

sthalik, 2015-11-01:
It works ingame? Looks like you fixed it some other way.

NaturalHate, 2015-11-01:
I was literally testing it before I restarted the client and before I tested the axis input method and it wasn't reacting, like usual. I'm not really sure what I did to fix it, but breaking it was almost as weird as this now ...

I know I started OpenTrack after ED the first time but I tried it the other way around before that too without success and I know I was able to start OT after Elite while it was working to activate head tracking too. Aside from that I just changed the protocol back and forth from Freetrack -> VJoy -> Freetrack.

What I notice now though is that I see the TrackIR.exe again in the task manager, I think (like 90% sure) that wasn't the case when it wasn't working (that was with the older version though so that might be unrelated or too old to be relevant). If I somehow bork it up again I'll have a look at that again though.

sthalik, 2015-11-01:
Reboot and try again as before. Is ED running as your typical user? Does it run as admin by itself?

NaturalHate, 2015-11-01:
Alright.  
And ED is running normally without admin rights.

NaturalHate, 2015-11-01:
Yup, still works after a reboot ...
That kinda sucks now for finding the issue though.

Are there still other people reporting this issue?
I don't really see anything in the forums at least.

sthalik, 2015-11-01:
FWIW here's an updated build https://db.tt/iHJ3uGlz

It might be an issue with mapping permissions, given you mentioned order of programs' start.

NaturalHate, 2015-11-01:
Kinda missed to test that with the build before but this one definitely works when started after ED too.

sthalik, 2015-11-01:
If someone still has the issue on ED forum please link them the build, thanks

bodylinksystems, 2015-11-10:
I've had the phone tracking/ip data  solution working fine in x-plane for OSX but I cannot gt it to work at all on the PC. I have to use my LED setup instead.  Any thoughts?

sthalik, 2015-11-10:
Add firewall exception explicitly for the port.

bodylinksystems, 2015-11-11:
will do. pc is a PITA. used to osx now.  will report back

sthalik, 2015-12-07:
@bodylinksystems any updates?

bodylinksystems, 2015-12-31:
sorry. no joy using the IMU version of the program on PC. Mac still ok.

been focusing on using the led version eclusively in DCS World, and would be happy to test the IMU part because I got a colorcross and would like to test VR capabilities.

MadMajkl, 2019-01-05:
Experiencing the same problem, don´t know what to do. Octopus moving while ED cockpit view fixed and ignoring movements :( ...

sthalik, 2019-01-06:
See https://github.com/opentrack/opentrack/issues/261#issuecomment-151403835

MadMajkl, 2019-01-06:
Got it working but in other way.... clear install of opentrack. No idea why it has not been working before but In first attempt I tried to use old config/profile files thats maybe the reason.

cashew12, 2021-12-17:
Sorry to necro, but i do have the same problem. Opentrack is installed on my F: Drive as is ED. It works with everything but ED.
I've tried the solutions given so far, but nothing.

sthalik, 2021-12-17:
Working around ED's mess is now more comfortable in newer opentrack versions.

[image]

Set the location to ED's install directory.

gufo2k20, 2022-06-15:
Same for me, :(
version 2022.1.1.....tried some trick (dll, location) but still not working.