opentrack/opentrack #262: opentrack 2.3 rc19p3 - Aruco Axis are broken!

JoeSchr, 2015-10-26:
Just tried the aruco tracker, came upon this bug: 
- Pitch gets translated as yaw, yaw gets translated as pitch. 

If I just go into option and map yaw to pitch and vice versa, everything works as expect. Kinda weird though.

sthalik, 2015-10-27:
What's your roll raw value?

JoeSchr, 2015-11-03:
I don't quite understand your question. or what data you require. generally roll seem to act fine. 

settings are: 
- yaw -> pitch [X] invert
- pitch -> yaw [ ] invert

[image]

does this screenshot help? 
[image]

this are the current settings:

sthalik, 2015-11-03:
Yaw and pitch are swapped since your model is rotated by 90 degrees. It shouldn't say roll: 86 by any means.

JoeSchr, 2015-11-03:
Thx, that really did the trick. Had to roate it 90° ccw. 

Maybe this should be advised in the wiki (because I printed it with the same orientation as linked here: https://raw.githubusercontent.com/opentrack/opentrack/unstable/contrib/aruco/test3.jpg )

closed 2015-11-04 by sthalik

sthalik, 2015-11-04:
> Maybe this should be advised in the wiki

If it really isn't obvious that the marker has to be rotated -- wiki is editable by everyone.

JoeSchr, 2015-11-04:
not obvious, edited wiki! :)