opentrack/opentrack #263: Oculus Runtime 0.8

gagagu, 2015-10-28:
Hi,
opentrack is not working with the actual oculus rift runtime 0.8. May be it is possible for you to update it?

Thanks forward for your work!

sthalik, 2015-10-28:
yeah, I'm fixing 0.2.5 for dk1 first so wait up.

sthalik, 2015-10-29:
Do we drop 0.4 or is it still necessary for some dk2 users?

gagagu, 2015-10-29:
Hi, i don't understand what you mean with 0.4, do you mean the oculus runtime version?
If yes, the most dk2 user are using 0.6 and above.

sthalik, 2015-10-29:
@gagagu please confirm https://db.tt/MfcB0MGS works

sthalik, 2015-10-29:
@gagagu https://db.tt/aoAlzG33

gagagu, 2015-10-29:
I've tested it out and get the following error "Unable to start Rift tracker". I'm using Windows 10 and started opentrack with and without "as Administrator". Both are not working.
I've tested the "Oculus Rift runtime 0.8.0 -- HMD" and "Oculus Rift DK2 -- HMD" tracker.

sthalik, 2015-10-29:
@gagagu i forgot to initialize, see https://db.tt/KZuofVpy

gagagu, 2015-10-29:
It works!!! Big thx for your work!

Tank939, 2016-03-24:
Hi im having the same problem and the link gives a 404, could i possibly have the patch as well please, thanks.

sthalik, 2016-03-24:
@Tank939 the changes were merged the last year.

sthalik, 2016-03-25:
@Tank939 can you run other Rift software with the newest SDK? What's your device model?

Rudyboi313, 2016-03-26:
I am having the same trouble, "unable to start tracker" running on sdk 8

Rudyboi313, 2016-03-26:
@Tank939 @sthalik

mm0zct, 2016-03-27:
Is it worth investing effort just now in 0.8 when 1.3 will be released in a couple of days?

As a kickstarter backer I should be getting a CV1 soon, so I'll try and make some time to integrate it. Sorry I've been pretty MIA since the initial plugin - life rather caught up with me for a while.

hcloud00, 2016-03-29:
Hi, I have the same issue unable to start rift tracker

Rudyboi313, 2016-03-29:
1.3 doesnt support the dk2 which is kind of obvious but maybe if you could help out thw poor boys with the dk2 thatd be great.

hcloud00, 2016-03-29:
how do i help out? it would be cool to see this work

sthalik, 2016-03-30:
Give it time. I'm unable to work on the project for the time being.

mm0zct, 2016-03-30:
1.3 does support the DK2, just not completely, and I think there are a few issues, but there has been much success with DK2 users playing games in the Oculus store now. I've not fired up my DK1 but I hear problems on the Oculus forums about it not starting games due to not having a tracker, or not displaying to the screen. For DK1's sake, and compatibility with older demos and games, I'll probably investigate having SDK0.8 and SDK1.3 plugins in OpenTrack. Currently however, my home PC room/office is still full of kitchen stuff, due to house renovations. I'll get working on it after my CV1 arrives (whenever Euro kickstarters ship) and home life settles down a bit. you shouldn't be waiting too long.

Rudyboi313, 2016-03-30:
Coo coo thanks guys much appreciated

sthalik, 2016-03-30:
@mm0zct check https://github.com/opentrack/LibOVR/tree/branch-080 for work already done

Lagahan, 2016-04-15:
Oh wow, didn't expect the support to be so up to date for this already, haven't used OpenTrack since the DK1 days! Did the CV1 come yet? I dug out GTR2 lately and it seems to work with VorpX, I used to use TriDef with OpenTrack back on the DK1. Pitty 1.3 doesn't have backwards compatibility for 0.8.

sthalik, 2016-06-16:
Is there anything missing other than 1.4.0 support?

MaximBesters, 2016-08-09:
Hi there: i'm having same issues as posted above: running opentrack 2.3 in windows 10 with a oculus DK2:  i get the error message: 'unable to start Rift tracker'... What can I do to solve the problem? with regards, Maxim Besters

sthalik, 2016-08-10:
See https://www.dropbox.com/s/zcxqqqczx3qpghu/opentrack-fixes-20160810_1.7z

It won't fix the issue, but spew out additional info why it fails. There's a message and an error code. I need both.

sthalik, 2016-09-21:
There were no changes to OVR 0.8 that could break it. Since no one provided the debug info I can't help any further.

closed 2016-09-21 by sthalik

andrewmyhre, 2016-10-26:
I'm having the same issue and I'd be happy to help with debugging.