opentrack/opentrack #265: [UX] Add clarification on the horizontal and vertical line of the graph

MathijsG, 2015-11-03:
It's very minor, but I think it will enhance the user friendliness:

Explain which axis is which, like this:

[image]

Excuse my paint skills, and just random naming, but these little clarifications will all contribute to easier use.

sthalik, 2015-11-03:
will do. good idea.

FlyingCircus-, 2015-11-04:
I'd rather not start new topics so let me drop this here.

@sthalik I know you like to facilitate user interaction and I’ve felt your pain every time you had to sacrifice simplicity to include new features. I believe interface design is very relevant in order to make a piece of software understandable, so here’s my feedback about what I think can be improved.

Here’s our current interface:

[image]

I find the current camera feed area quite large for what I use it for, which is just setting the minimum and maximum dot detection size in point tracker and setting the threshold until I distinguish the dots from the background. That operation might be confusing for new users (and might require a wizard in the future) but it’s actually quite straightforward. On the other hand, I find the octopus picture feed area quite small, while I had to spent more time tweaking the mapping curves and setting the filters than in anything else.

I also find the location of the different buttons not very hierarchically placed. It’s difficult to see that each profile determines the tracker, protocol, filter and mapping curves assigned. The start/stop buttons in the middle seem also out of place. Finally, I like symmetry, hence I propose something like this:

[image]

The whole window has the same height than the current interface but it has a 15% smaller width.
The center placement of the profile selector is chosen to illustrate which settings are linked to it. I also changed the “mapping” label to “output” and added the (…) button just to provide a way to launch the “output” tab from there instead of options, so those settings (cross axis settings, axes inversion, translation compensation…) are also linked to the profile (which I think makes sense).

On the other hand, by having two equal sized areas for the camera feed and octopus feed we can place the raw axes coordinates and game data coordinates below each one and have a clear parallelism between what the camera sees and what the software translates to.

MathijsG, 2015-11-05:
Still find it very cluttered. when I have the time I will think about some UI proposals.

I personally find the octopus not giving a good representation for 3D space. perspective is sometimes hard to distuingish. Can't we switch to a 3D model either? Like FreeTrack had and TrackIR has.

sniperwolfpk5, 2015-11-05:
@MartinJG Well A 3D model will eat resources. I would suggest a toggle option for either Octopus or a 3D skull

sthalik, 2015-11-05:
There were issues with running an opengl widget at the same time as games.

There won't be a toggle option, whichever option is used.

Would it help if the model was projected with perspective and not orthogonally?

MathijsG, 2015-11-05:
Maybe, could you when you have the time, provide a build with perspective?

And maybe we can use a scalable vector image of the octopus? So it doesn't distort ugly?

FlyingCircus-, 2015-11-05:
I believe a perspective view would cause more confusion than an orthogonal one. I agree that freetrack and opentrack had a better way to represent your current pose. Personally I wouldn't mind a toggle button in case we had different visualizations, but toggle or not the hard part is achieving those visualizations. Is there a way to fake a 3d model with no opengl involved? I imagine for example we could superpose a layer with the facial features (eyes, mouth) subjected to current perspective transformations over a layer with a simple 2D sphere-like shape subjected to only translation transformations... I should draw that to illustrate what I'm talking about. Although we would lose the octopus we've learnt to love.

MathijsG, 2015-11-05:
I could model an Octopus :-). 

I mostly have problems with representations like this:
[image]

The regular straight up and down and right and left are all very clear.

MathijsG, 2015-11-05:
Some other suggestions:

- When actively moving a point in the mapping graph, show a label somewhere (or maybe right next to the position of the dot) with the current value of that dot, this makes it much more accurate to get specific degrees, like 90 degrees and stuff.

- Give the point optional bezier handles to change curvature easily

sniperwolfpk5, 2015-11-05:
@MathijsG I agree with your suggestions. At least the first point is more achievable

sthalik, 2015-11-05:
@FlyingCircus- It could be a modelled octopus ellipsoid head also. The real issue with 3D rendering is knowing what's occluded and keeping vertices from overlapping based on depth.

If you can 3D model the octopus - maybe even more cute and absurd in the place like this one - that'd be fucking great.

@MathijsG 
> When actively moving a point in the mapping graph, show a label somewhere (or maybe right next to the position of the dot) with the current value of that dot, this makes it much more accurate to get specific degrees, like 90 degrees and stuff.

Fucking amazing idea, no sarcasm. Actually we should snap to 0.5 degree when dragging.

> Give the point optional bezier handles to change curvature easily

What? We're using Catmull-Rom splines and not Bezier.

MathijsG, 2015-11-05:
The current octopus is a vector on the internet I think? We can save that as a big rasterized image maybe?

About the curves, I had no idea. I do often work in programs where you have these curves which you can control with handles. :-)

sthalik, 2015-11-05:
We can but  that grows the executable non-trivially. Older trackhat versions had a few-meg executable cause of different-resolution icons lol. Submit a pull request with 256x256 octopus if you wish. Just compress the squid out of it.

Force snap to 1deg? I'll make a build, as that makes sense. There won't be checkboxes for every single feature though :)

sthalik, 2015-11-06:
@MathijsG drop me a model in OBJ format please.

MathijsG, 2015-11-06:
I will try modeling when I have the time. Can we use a placeholder 3D model for testing purposes? There are tons of free heads, or even free octopus models out there.

MathijsG, 2015-11-07:
Try this: https://mega.nz/#!aYxlkYAA I made some simple modifications to an existing model, for testing purposes: https://mega.nz/#!aYxlkYAA

sthalik, 2015-11-08:
Asks for a decryption key, please host it somewhere normal...

MathijsG, 2015-11-08:
Woops then i didnt copy the full link.

Will fix later today.

MathijsG, 2015-11-09:
https://mega.nz/#!aYxlkYAA!TPtSQJCJahCTUs-yWBQ5TPKpTfeBIat0jnaTdoIPds8

sthalik, 2015-11-10:
Please upload the texture `2274708_Blue ring octopus\mesh\Creature.png` as well.

sthalik, 2015-11-10:
I'm so into this. octopus will be lighted from the point of origin. prepare for absolute ambient/diffuse/specular terror.

MathijsG, 2015-11-10:
Here are the original files, I didn't stumble upon that one: http://storage.123dapp.com/Download/index.php?ID=Z1556930&afc=PA1ON1&session=5FC535FB-C3AC-4392-A784-14B021D3EDD7

sniperwolfpk5, 2015-11-11:
Looking forward to monster Octopus :)

MathijsG, 2015-11-11:
Maybe only let the model be viewed when it is the active window? So it won't cost any resources while gaming?

sthalik, 2015-11-12:
Inactive windows can be visible too. Let's leave it to Qt's logic.

sniperwolfpk5, 2015-11-12:
What about minimize to tray?

sthalik, 2015-11-12:
We'll see how many resources it takes first. It has to take very little anyway. Then it'll be a moot point when to disable it.

sthalik, 2015-12-18:
@MathijsG I actually like your idea of making the interface more compact. Sadly there's no 3d octopus yet but it gained transparency powers. See the latest patchlevel.

sthalik, 2015-12-19:
I'm closing this, I have no time for making an OpenGL octopus. Many things can go wrong and even if it works, the improvement is pretty minor. I'll still accept third-party code though.

closed 2015-12-19 by sthalik

MathijsG, 2015-12-19:
Too bad, but I understand :-)

MathijsG, 2015-12-22:
But what about the other improvement stated in my initial post? The very easy explanation of the horizontal and vertical axes? And maybe the hover value when you move with the cursor over the curve?

sthalik, 2015-12-23:
@MathijsG the hover value feature is rendered moot by snapping X and Y values on the graph to a grid.

We're working on a "quick start" wiki page describing the mapping window among other things.

The problem at this point is lack of documentation, see #277.

MathijsG, 2015-12-24:
Can we instead, or as an addition to the wiki make the fieldsets with the names "Protocol" "Tracker" "Filter" and such clickable with some popup window or something? So people get the info right in the application instead of searching for a wiki in this github.

sthalik, 2015-12-24:
That's gonna make it royally annoying when clicking the links by accident while reaching for the comboboxes.

sthalik, 2015-12-24:
We're gonna have a quick start guide, giving the user all he needs to know. That should suffice.

I'm gonna edit the wiki, but needs a few days/weeks of rest, depending on how things turn out.

MathijsG, 2016-05-24:
How can I pull the octopus like you said in: https://github.com/opentrack/opentrack/issues/265#issuecomment-154155486

I want to contribute my part, but I can't code, so I maybe can improve this.

sthalik, 2016-05-24:
@MathijsG it's here https://github.com/opentrack/opentrack/blob/unstable/gui/images/facetracknoir.png

MathijsG, 2016-05-24:
Here I have the compressed but bigger octopus. Made in PNG-8

[image]

sthalik, 2016-05-25:
@MathijsG thanks, I'll commit that.

Did you know that with Qt Designer you can change the UI without programming knowledge? You can fix user experience issues that way. Just saying.

MathijsG, 2016-05-25:
Nice, I will look into that later :-).