opentrack/opentrack #267: Joystick input forgets selected joystick

runningman84, 2015-11-15:
I have two ore more joysticks connected to my machine. opentrack-2.3 rc21 forgets the selected joystick once I stop the tracking and open the tracker settings once again. So each time I open tracker settings I have to change the selected joystick to the right one (edtracker instead of x65) again.

sthalik, 2015-11-16:
What's your .ini value for `joy-guid`?

sthalik, 2015-11-16:
Is it consistent between runs?

sthalik, 2015-11-16:
Confirm works https://db.tt/r2dlFKZh, thanks

runningman84, 2015-11-16:
The version works but if I click on stop the application freezes.

sthalik, 2015-11-21:
it's in progress. we can apparently only have one joystick handle for the whole application at a time so a little refactoring is in order.

sthalik, 2015-11-21:
that is, joystick tracker will have to use core opentrack functionality that needs to exist to begin with.

sthalik, 2015-11-22:
Please confirm if this build works as advertised. In particular, I need to know whether multiple joystick selection is broken. I got only one stick so couldn't check it. @Len62 can you help?

The build - https://db.tt/1bXkwThn

sthalik, 2015-11-22:
Here's the same thing but without debug info. Smaller that way.https://db.tt/2DjeFOfO

Len62, 2015-11-22:
D/L the w/o debug install. I was able to assign buttons both from my joystick and throttle, which are separate USB devices. Settings were remembered by opentrack, as I started/stopped, open/closed the program multiple times. Did not experience any issues.

sthalik, 2015-11-22:
@Len62 thank you. I'm releasing rc21p1.

sthalik, 2015-11-22:
The joystick tracker should now also work when hotplugging/unplugging the stick's USB connector. I think it shouldn't crash on unplug during tracking in any cases. Report any issues.

runningman84, 2015-11-23:
The new version seems to work but the latency is really high and the movement is lags...

2.3rc20 works a lot faster.

sthalik, 2015-11-24:
@runningman84 is it for all trackers or only joystick input?

runningman84, 2015-11-24:
I do not have any other tracker

sthalik, 2015-11-25:
@runningman84 confirm this build works. There was this minor issue https://db.tt/dBrPzRWf

runningman84, 2015-11-27:
no opentrack-2.3-rc21p1-1-gf09ff29-20151125022606 suffers from the same problem...

sthalik, 2015-11-29:
Compare if https://db.tt/x06WJYor is any better, thanks.

sthalik, 2015-11-29:
oops

sthalik, 2015-11-29:
Use this one instead - https://db.tt/PW4OmoMd

sthalik, 2015-12-02:
@runningman84 is it any better in rc21p2?

runningman84, 2015-12-02:
I will report this night, sorry for the delay...

runningman84, 2015-12-02:
still very bumpy movement with opentrack-2.3-rc21p1-12-ga07eed2-20151129123438

sthalik, 2015-12-05:
@runningman84 I checked yaw using a throttle and there was no jerky movement. There were a few crashes in p2 that were fixed since then. Provide more info:

- is there low CPU usage as it should, or is it consuming one core or something
- is it also lagging with a stick/throttle as a controller

runningman84, 2015-12-05:
Just tested 2.3-rc21p3. The cpu usuage is about 3%. Yes it is also lagging using my x65 stick.

Funny enough the old release still works fine. I am running windows 10 x64. I can record a desktop view if this helps you?

sthalik, 2015-12-06:
How many USB joysticks do you have in total in the selection box?

Does disabling filtering help change anything?

How do you describe the erratic movement - is it pausing for fractions of a second while tracking or something else?

sthalik, 2015-12-06:
The old build didn't query for additional joysticks being plugged in or out. This is likely the cause.

sthalik, 2015-12-06:
I managed to remove the need for synchronization in most of the code. In particular, joystick tracker and joystick keybindings aren't using the same data anymore. This should help you out. Initial reports show no crashes or freezes.

If it's still stuttering, I'll remove refreshing the joystick list from asking for axis values.

@runningman84 build - https://db.tt/fvbAGatr

runningman84, 2015-12-06:
The latest build doesn't help either.

The movement is slow and doesn't look smooth.

I have a joystick and edtracker.

sthalik, 2015-12-06:
@runningman84 I'm not polling for new joysticks in this build - https://db.tt/tUBGEgRj

Any luck?

runningman84, 2015-12-06:
yes this build is working fine!

sthalik, 2015-12-17:
#279 is the last one left. I'm closing this.

closed 2015-12-17 by sthalik