opentrack/opentrack #269: Double/triple vision when using the PS3 eye

heikkisiltala, 2015-11-19:
Hi.

I'm quite new to opentrack. Previously been using freetrack, but decided to migrate to opentrack since FT has problems with new games. I just bought a new PS3 eye camera and ran into a problem:

Image shows like it should on CL-test:
https://drive.google.com/file/d/0B3zOqOKL82TCc180bE4yTk14bFU/view?usp=sharing

But on opentrack it's like this:
https://drive.google.com/file/d/0B3zOqOKL82TCOUl6TmZORVN3NnM/view?usp=sharing

Camera settings in opentrack:
https://drive.google.com/file/d/0B3zOqOKL82TCaUZUTzhoN0I5bnc/view?usp=sharing

Any idea what is going on?

sthalik, 2015-11-19:
Either connect to a usb 2.0/3.0 port or lower framerate.

closed 2015-11-19 by sthalik

heikkisiltala, 2015-11-19:
Hi.

I made sure that the camera is connected to a USB3.0 port, no joy.
I then lowered the resolution to 320x240 and fps first to 30... then to 10. No joy.

sniperwolfpk5, 2015-11-20:
Re-install PS3 eye drivers

heikkisiltala, 2015-11-20:
But but but... the CL-eye test image is fine? ... Well, I'll try when I get home from work.

sniperwolfpk5, 2015-11-20:
I think it is issue with you specifically. As I am using last release version with PS3 eye and I am facing no issue. What comes in my mind are
Play with Point Tracker settings. 
Make sure you have selected PS3 eye from camera list. 
Another think try to use resolutions that are displayed in CL-eye.
Try to run as administrator
Check your modal and put correct measurements

sthalik, 2015-11-20:
I'm positive there's an issue with your USB hub. Image gets corrupted in the form of displaced frames where USB packets are dropped.

heikkisiltala, 2015-11-20:
I have no USB hub, the external kind, if you mean that. The camera is directly connected to the back of my desktop computer (mobo: Asus M4A87TD/USB3) to an blue port which manual says is a 3.0 port.

But, now leaving work, I will test that driver reinstall when I get home.

heikkisiltala, 2015-11-20:
Uninstalled the driver, restarted, downloaded the driver again just to be sure, restarted again. Again CL test shows good image, but opentrack not so much :(

I tried with 640x480 60fps, and 320x240 30fps.

sniperwolfpk5, 2015-11-21:
If CL test shows good image. Then there is no issue with your USB port but I think there is another camera application installed that is sharing or interrupting. Do you have any other camera app installed other than CL and opentrack? Because i faced issue once but that was not distortion rather my ps3 eye was not able to give me 60fps.

sthalik, 2015-11-21:
CL-Eye defaults to 30 fps, select a different pin for the camera.

heikkisiltala, 2015-11-22:
What could this interfering camera app be? I have several camera using apps. Several opentrack versions, FreeTrack 2.2, FTNoIR.

I do have my former camera's (MS VX-3000) driver installed, should I uninstall it? The camera itself hasn't been connected.

heikkisiltala, 2015-11-22:
Different "pin"? This is a foreign term to me. Do you mean a different resolution / fps?

sthalik, 2015-11-23:
@heikkisiltala select different "pin" option in CL-Eye. Select 640x480 and 75 Hz. Does the issue occur in CL-Eye on this high framerate then?

sthalik, 2015-11-23:
@heikkisiltala try a different USB port as per its color. Some ports may be black, others blue. Look inside the port on the PC and try all the distinct colors you have.

heikkisiltala, 2015-11-23:
The CL-Eye test image was all good on "640 x 480 px (32 bits) @  75.000 FPS" ... Same on 24bits. I will try to cycle through the different ports tomorrow, although I have already tested one blue (2xUSB 3.0), and one black port (6xUSB 2.0).