opentrack/opentrack #271: Tracking lags a lot when recording/streaming with any software

gienkov, 2015-11-26:
As stated in title, I experience very annoying issue with OT. Basically I cannot record or stream when I have tracking active. Game FPS are ok but tracker jitters to a point where I start to have headache.
You can see what I mean on my yesterday stream here:
https://youtu.be/CvBvlpotdgo?t=81

sthalik, 2015-11-29:
How many CPU cores? How many cores encoding takes? Tracker used?

gienkov, 2015-11-29:
4 cores, point tracker, x264 encoder.

sthalik, 2015-11-29:
How many cores does ED use? How many does encoding use? Can this be solved by adjusting opentrack priority to realtime?

sthalik, 2015-11-30:
@gienkov update to rc21p2. There was a runaway thread consuming one core since p19, now fixed.

gienkov, 2015-11-30:
Zabieram się za testy ;)

gienkov, 2015-12-01:
It seems somewhat better now, but jitter is still existent. I have way more smooth input when I'm not recording than when I do.

sthalik, 2015-12-02:
Set opentrack to realtime priority in Windows task manager. There are likely not enough cycles for it.

koreldan, 2015-12-02:
I've had the same problem.
I solved changing filter from "Accela" to "EWMA"

I use OpenTrack with the EDTracker Pro & OpenTrack v2.3 rc21p2

gienkov, 2015-12-02:
I'l try EWMA then, testing in few mins

HornetMaX, 2015-12-02:
I have an EDTracker Pro too and I don't need any filter at all in OpenTrack. What's the need ?

koreldan, 2015-12-02:
It flicker a little to me, maybe because it has too much sensitivity... dunno

i've set "pitch scaling" and "yaw scaling" to 1, "smoothing" to 99 and "response mode" to linear.

Any tips ?

HornetMaX, 2015-12-02:
Weird. I have mine with pitch and yaw scaling at 1, smoothing at 50 and no filter in opentrack, zero problems.

koreldan, 2015-12-02:
Maybe i should re-calibrate the magnetometer ?

HornetMaX, 2015-12-02:
Could help. Also remember to keep the EDTracker still the first 20sec you power it up, as it does a bias compensation each time.

koreldan, 2015-12-02:
sometimes after the 20sec i click on "auto bias" to be sure that it will be somehow more stable

HornetMaX, 2015-12-02:
Same here, I just hate the auto auto-bias at startup.

koreldan, 2015-12-02:
I was used to use a self-made headtracker with 4 leds hat + a crappy webcam... it worked pretty well but the "EDTracker Pro" experience is pure gold to me ;)

HornetMaX, 2015-12-02:
Yeah, EDtracker rocks. I'm really waiting for their wireless version. OK, back on topic now :)

koreldan, 2015-12-02:
I've just realized that you are the author of OpenTrack... awesome work, man !

HornetMaX, 2015-12-02:
Me ? I'm definitely not ! sthalik is.
I didn't even contribute a single line :)

sthalik, 2015-12-02:
If it works in EWMA but jumps around in Accela, it's definitely a but in Accela. We should get to the bottom of this rather than to prescribe workarounds.

HornetMaX, 2015-12-02:
Sure, I agree. Was just surprised of the need of a filter with an EDTracker.
Of course there are other devices that do need a filter, so it should work (or be made to).

koreldan, 2015-12-03:
it doesn't jump around.... it flickers a little with the accela filter.

Anyway, I've removed also the filters and re-calibrated the tracker and followed some suggestion provided by HornetMaX.

sthalik, I'm sorry for my wrong attribution of the project... and thank you for your effort on keeping the software working. great work  ! ;)

sthalik, 2015-12-14:
To diagnose the issue with Accela, I already asked you to adjust process priority and see if it still occurs. I can't help you if you won't help me diagnose the issue.

koreldan, 2015-12-14:
sorry... I'll try tomorrow

sthalik, 2015-12-14:
If issue still occurs with realtime priority, compare game fps when encoding and not encoding.

koreldan, 2015-12-15:
ok.
Acella seems fixed with version opentrack-2.3 rc21p4

I've also tried with opentrack-2.3 rc21p2 ... that version still has problems with acella.

By the way... last release fixed acella problems... no process changes needed ;)

sthalik, 2015-12-16:
@gienkov are there any issues left here?

sthalik, 2015-12-17:
I'm closing this. Please comment if there are any more issues.

closed 2015-12-17 by sthalik