opentrack/opentrack #272: OT 2.3 RC21 Crippling System Interupts

0ddBall, 2015-11-28:
I have revered back to RC20 and issue does not appear there. When tracking and playing war thunder, system interrupts run excessively and kill frame, ping and the system in general because unresponsive but does not crash.

[OpenTrackBugReport.txt](https://github.com/opentrack/opentrack/files/46289/OpenTrackBugReport.txt)

sthalik, 2015-11-28:
What system interrupts? Are you talking about page faults?

What tracker are you using?

0ddBall, 2015-11-28:
System Interrupts in my brain, please disregard

closed 2015-11-28 by 0ddBall

sthalik, 2015-12-05:
@0ddBall you might mean the multimedia timer. opentrack sets OS scheduler Hz to 1000 during tracking using timeBeginPeriod and restores it back to 100 once it's done. It shouldn't have any negative effects on modern PCs. My 486 Geode runs on 500 Hz permanently without issues.