opentrack/opentrack #273: aruco hight cpu usage

frost555, 2015-11-29:
Hello i'm using aruco tracker (640x480 60fps) and CPU load from opentrack sometimes goes up to 45% on my core i5 2500k system. Is it normal?

frost555, 2015-11-29:
[image]
[image]

sthalik, 2015-11-29:
It's only expected to do higher if there's no detection each frame.

sthalik, 2015-11-29:
To elaborate - in case there's no detection for 1.2 sec, aruco will cycle to a different detection value suitable for another light level. If there's more likely a detection than not, aruco will persist at that detection value.

Do you get flicker with the pink rectangle and all?

frost555, 2015-11-29:
Nope. Tracking is stable.

sthalik, 2015-11-29:
Does the issue persist with protocols other than SimConnect?

frost555, 2015-11-29:
I played Assetto Corsa with freetrack protocol for 30 minutes - and CPU usage got high to 30%. Even if i stop playing and just sit in front of camera - cpu usage does not go down

sthalik, 2015-11-29:
Does it also happen in rc20?

The issue is likely in another component since Aruco runs single-threaded. It should use not more than 25% of a quad-core CPU.

sthalik, 2015-11-29:
Before downgrading to rc20 check https://db.tt/PW4OmoMd first, please.

frost555, 2015-11-29:
Seems better with last commmit. CPU usage didn't got highter than 15-16%.

frost555, 2015-11-29:
https://goo.gl/photos/5hnwrevhSNLvpzuh9 here is the video. It seems like something is flipping before losing tracking, maybe it is the cause of cpu usage

frost555, 2015-11-29:
Also i'm using marker from generated from this site (id: 1) http://terpconnect.umd.edu/~jwelsh12/enes100/markergen.html
Could not run generator provided with opentrack. Dont know if it cause problem.

sthalik, 2015-11-29:
See my description of the new changes <<https://github.com/opentrack/opentrack/commit/e1ae4a738c011fbbe44e5203cc304dacb6489768>> along with earlier description of threshold cycling from here.

You can run the marker generator from Windows cmdline as long as the generated file extension is .bmp. It doesn't have the support for other file formats.

Your marker selection is great actually, should be extremely easy for the software to detect. It's recognized and detects each frame.

New build with the changes - https://db.tt/kdCFWOHk

The 50% cpu usage thing looks like the joystick support regression, that's fixed after rc21p1.

sthalik, 2015-11-29:
For the actual diagnosis of your high CPU usage - cover the marker for over 0.35 seconds but not more than twice than that. See if it gets cycled back and forth between high CPU usage. Information about cycling gets shown only on the program's console, sadly.

If you have MS Visual C++ (accessible for separate download since recently) you can access "editbin /SUBSYSTEM:CONSOLE opentrack.exe" to make the console pop up. Alterlatively you can launch the software from a Cygwin terminal or any debugger.

frost555, 2015-12-01:
I have attached debugger. CPU usage sometimes goes high after losing tracking. Message appears "otsu now off" and it didn't recover from this state until i lose tracking again. Sometimes it goes like "otsu now off" and then "otsu now on". But sometimes just "otsu now off"

sthalik, 2015-12-02:
Does a box size chosen correlate with CPU usage? Box sizes 5 and 11 are the most extreme. In addition to otsu messages you're getting messages like:

    aruco: otsu now true
    aruco: box size now 11
    aruco: otsu now false
    aruco: otsu now true
    aruco: box size now 5

I have a hypothesis that box size 5 + otsu off causes your high CPU load. That image is the most noisy. Can you confirm that it does that, and that higher box sizes don't cause higher load, regardless whether otsu thresholding is enabled or not?

frost555, 2015-12-02:
box size does not affect cpu load significantly. Only when last message is otsu now false - then cpu load goes high

sthalik, 2015-12-03:
I removed box size logic and enabled Otsu unconditionally. It works fine for my use cases.

Build: https://db.tt/M6aaQSuY

frost555, 2015-12-04:
Tried this build with great success - cpu usage never goes high unexpectedly. One more option i would like to ask - is an ability to regulate size of pink box. I think i can live with smaller area, especially with high fps mode (as i can move less distance within shorter frame) and it will lower the cpu usage. What do you think?

sthalik, 2015-12-04:
It has to work for 30 Hz, the lowest common denominator. I'm not entirely sure how much movement there is on a frame-by-frame basis. Could you do some tests for me please?

frost555, 2015-12-04:
In my brief tests i managed to move the marker by about 30px with fast head movements when marker size itself is about 130px. Video mode was 640x480 30fps.

frost555, 2015-12-04:
https://goo.gl/photos/rVjbgN8qitC4CYBr8 video 
and my diletant measurements 
[image]
[image]

frost555, 2015-12-04:
Right now pink area would allow for about 90pixels move per frame. (If i did my calculations correctly :)

frost555, 2015-12-05:
Tried new release and had no problems. Cpu load is about 1.5 time lower. Managed to break tracking only on 30fps with very rapid translation movement. Also it seems that disabling box size logic and otsu switching made tracking more stable. Because i could not manage to get rc20 aruco tracking work stable with daylighting. And rc21p3 tracking is stable.

sthalik, 2015-12-05:
Looks like we're done here, thanks for confirmation.

closed 2015-12-05 by sthalik

sthalik, 2015-12-05:
@frost555 benchmarking is tricky, since CPU load depends on closeness of the marker to the camera. Bigger it is, bigger the area to process. Also noise levels due to gain+light change things a bit.