opentrack/opentrack #274: Opentrack crashing repeatedly

mursey, 2015-12-03:
I'm trying to get opentrack working with ED Headtracker and ED again, but it keep crashing out after working for a few minutes. I've got it extracted into the same directory as ED, all permissions are checked for everyone and I'm on windows 10 64. No idea what's causing it to crash, it just does.

mursey, 2015-12-03:
Nevermind, I fixed it. Just need to make sure freetrack 2 is using the 64 client.dll

sthalik, 2015-12-03:
What .dll was it using before?

mursey, 2015-12-03:
It was just using the NPclient.dll originally.

Also the crashing came back, so I fixed it from not working with ED in game, but now I just need to keep opentrack from dying during a firefight.

I can provide any info you need later on today, but I leave for work soon and can't access my computer until after.

sthalik, 2015-12-03:
Looks like there's one more bug in new joystick support. Are you using rc21p2 though?

mursey, 2015-12-03:
I am. I was using 2.2 installed to an old hdd, but I upgraded to the latest  2.3 and put it on my ssd in the frontier program x86 parent folder.

sthalik, 2015-12-03:
Hey, here's a new build. if #275 and #274 confirm it's working it's gonna be patchlevel 3.

Build: https://db.tt/UthSld0u

mursey, 2015-12-03:
Just got home. I'll download, calibrate my ed HT, and report back.

mursey, 2015-12-03:
So, still getting crashes. If I use NPClient.dll it's pretty much as soon as I start ED (doesn't crash if I have it going and am not playing anything). If I use NPClient64.dll, I get like 2-5 minutes, and then it crashes.

sthalik, 2015-12-04:
Does this happen on different trackers? Can you use an irrelevant tracker just to see if it crashes?

mursey, 2015-12-04:
It does. I tried vjoy and it crashed

sthalik, 2015-12-04:
Try this:

- reboot
- start opentrack before other tracking software
- start tracking in opentrack
- close it, start everything as usual

Will you get the crash then?

sthalik, 2015-12-05:
@mursey rc21p3 got released and few other users report that crashes are gone. Try this one instead.

mursey, 2015-12-05:
@sthalik still getting crashes. Tried the above and the new version, no dice. I couldn't even make it into Elite Dangerous before it quit.

sthalik, 2015-12-06:
Sadly I need debug info.

- Install mingw-w64 from the online installer
- Open a Windows console prompt. Change directory to where opentrack is installed.
- Run "gdb.exe" from the mingw-w64 installation. You can open the folder graphically, then drag the gdb.exe icon to the Windows console prompt
- on gdb prompt, type "file opentrack.exe". It should run opentrack.
- wait for it to crash
- type "thread apply all bt"
- send this output

sthalik, 2015-12-06:
You need a build with debug info in it. Here's one - https://db.tt/r5Ve6vIr

mursey, 2015-12-06:
Uh, it's not running opentrack after I run it.
I've tried it several times, with no luck. Picture attached to show my steps

[image]

sthalik, 2015-12-06:
Sorry. After the "file" command is successful, type "run" by itself as the next command. opentrack will then start.

mursey, 2015-12-06:
There we go. Sorry, didn't know how to load it into a text file.

[image]
[image]
[image]
[image]

sthalik, 2015-12-06:
This one looks strange: `c:/program files (x86)/gigabyte oc_guru ii/GvKm.dll`.

It's using DirectInput while injected into the opentrack process. It's obviously calling for trouble. It's recreating the same scenario we had earlier in p1, but this time it's out of our control since it's a third-party DLL.

It can also explain why you're getting nonsense when polling keyboard in thread 7. It dies when getting a list of pressed keys.

Try to disable this .dll using "sysinternals autoruns" software.

mursey, 2015-12-06:
It's just my  980ti's oc software, so I guess I can try turning it off, even though it's overclocking my graphic card.

mursey, 2015-12-06:
I closed out OC Guru II, so far no hang ups

sthalik, 2015-12-06:
I have a theory why it's broken like that. Hold on.

sthalik, 2015-12-06:
@mursey Try this build with uncommitted changes - https://db.tt/DnVZN0KX

mursey, 2015-12-06:
testing

sthalik, 2015-12-06:
@mursey no need. There's no debug info in that binary. I think I know the solution[1] to this and jerky joystick input.

[1] register Direct Input handle with a fake invisible window

Should take me a few mins.

mursey, 2015-12-06:
take your time man. Also, do you have a donate page? I don't think I've seen a dev be this attentive to issues.

sthalik, 2015-12-06:
@mursey the rc21 release broke joystick support in general. We're getting 1.5k unique downloads a week so I'd like to avoid further alienating users with these few bugs. I hope things won't be as hectic as the last few days.

We don't take donations due to the "NPClient" murky legal territory. It's uncertain as it is. :(

Good news is I can't get it to crash anymore, with joystick tracker and joystick keystrokes. Have a try - https://db.tt/fvbAGatr

sthalik, 2015-12-06:
And keep the overclocking software running, opentrack definitely needs to run despite that.

mursey, 2015-12-06:
That's a shame. You guys really need something for all the work you do.

So far no crashes.

mursey, 2015-12-06:
Nevermind, still crashing with oc guru. I uninstalled it (didn't like it anyway) and started using msi afterburner. So far it's working.

sthalik, 2015-12-06:
@mursey do me a favor and confirm lack of freezes/crashes after few days of opentrack usage. Thanks!

sthalik, 2015-12-17:
@mursey any more issues? Please reopen if something bad happened.

Artazole, 2016-02-25:
Hello, 

may i re-open this thread just to say that i'm using EDTracker Pro, then OpenTrack to use with DCS or Assetto Corsa and even in Arma 3 but i have Oc'GURU and it took me time to found out that opentrack crashes comes from that software. 
Sometimes i don't run it, but for long time play, it's safer for my GPU.
I'm using 2.3-rc21p8, i have W7 Pro 64bits and GTX 970 Gigabyte.
Is there any way this could be solved ? I'd like to keep this fine software cuz it's simple and efficient and i'm not into overclocking just need that software to manage the fan % depending on GPU Temp.
Thanks anyway for all the hard work.

sthalik, 2016-02-25:
@Artazole the software

sthalik, 2016-02-25:
@Artazole the software might inject a .dll into the opentrack process. The "RadeonPro" software caused crashes all over due to a buggy injected .dll. This may be similar.

Look for OSD or keyboard functionality and disable it in the software, it might help.

Artazole, 2016-02-25:
Thanks for the FAST reply !! 
How can i "catch" this .dll you talking about ? Sorry, i have no idea about that... 
And whatever way opentrack and Oc'Guru are started, it crashes after a few second. 
Sorry for my incompetence.

sthalik, 2016-02-25:
Go to OC'Guru options and disable OSD and keyboard shortcut functionality.

Artazole, 2016-02-25:
I see what you mean.
I just look at it, its disabled...

So i try OpenTrack with OSD enable is case... still crashes. 
Don't want to bug you over, thanks for trying to help this late at night and so fast.
If you ever have an idea, i'll be all yours...
Bonne nuit ;)

sthalik, 2016-02-25:
http://forums.steampowered.com/forums/showthread.php?t=2075101

Replace it with MSI Afterburner or similar.

closed 2016-02-25 by sthalik

Artazole, 2016-02-25:
Hello back,

i think you're right, i'm gonna have to use something else.
Sorry for the inconvenience, and thanks again for your help trying to find out. 
Keep up !!!